ratgdo/esphome-ratgdo #266: Paired Devices Status and Openings count Incorrect?

hjdhjd, 2024-05-01:
From a fully factory reset Liftmaster garage door opener…I added five remotes to it in total.

In the myQ app, under myQ Diagnostics, it shows (correctly) 5 remotes (and 0 keypads).

In the Ratgdo ESPHome webUI I see: 7 paired devices.

Anecdotally, there’s also another discrepancy:

myQ Diagnostics shows 684 cycles.
Ratgdo shows: 680 openings.

Thoughts? In the grand scheme of things these are minor annoyances…just thought it would be helpful.

mwaterbu, 2024-05-16:
I wonder if the obstruction sensors count as paired devices? My count is also off from what I would expect.
I believe myQ connections also count here, based on #149.

restauffer, 2024-05-16:
I think the MyQ counts as a paired device in the Ratgdo. When you first get a myQ you pair it to your GDO similar to a remote so it makes sense it would appear as a paired device.  Possibly both the MyQ base and the sensor on the door count which would explain why you are off by two. Just my theory.

hjdhjd, 2024-05-16:
Possibly. 😄 Still, it would be strange since the GDO itself is counting itself as a registered remote.

What would be a nice bonus feature would be the ability to identify individual remotes in ratgdo. myQ can do this, these days, and so you can know XYZ just opened the garage, etc. There’s endless automation use cases there too for folks so inclined.

cwross, 2025-03-03:
I just set up a new RatGDO for the first time. I have 3 remotes, and I have never used the MyQ app or even connected the opener to Wi-Fi. RatGDO is reporting 4 remotes. I'm guessing (hoping) it is either the obstruction sensor or the wall control panel that accounts for the extra device.

It would be neat if those devices could somehow be enumerated in a useful fashion, but I'm sure that's easier said than done.