ratgdo/esphome-ratgdo #267: Light Status Occasionally Incorrect.

hjdhjd, 2024-05-01:
No rhyme or reason here…after a few days, inevitably, I will find that the light status is incorrectly reported in the Ratgdo webUI, showing that the light is on when it is in fact off.  If I force a sync or a restart, Ratgdo detects the correct state once more.

Don’t have a good path to recreate this one…what I can say is that we never manually turn on/off the light. The only times it comes on is when the GDO has decided it should come on (motion event, opener movement, etc.).

Happy to help troubleshoot if needed.

mwaterbu, 2024-05-15:
I have found this as well, but in my case, the incorrect state is only present for at most 5 minutes.
What I was finding was that the light would turn on from the motion sensor on the wired wall remote (and the status would show this right away), but when the wall remote eventually turned the light off after motion was no longer detected, the light would still show itself as being on in ratgdo. However, after a few minutes, the ratgdo status updated and showed that it had turned off.
This was an issue for me, as I was trying to turn on the light if a door into the garage was opened (even if no motion was detected), but if the light was already showing as on, nothing would happen. Eventually, I found that @PaulWieland [mentioned](https://github.com/ratgdo/esphome-ratgdo/issues/174#issuecomment-1892454384) that the Security 2.0 openers send out their status only every 5 minutes, which seemed to be exactly what I was noticing. The status is sent out when the light turns on from motion, but isn't sent when it turns back off, so you just have to wait for the next 5 minute update.
The solution for my automation was to call a service to press the "Query status" button prior to turning the light on. This allows me to get the correct entity state if the automation fires in the window of time between when the light turned off and the next 5 minute update.

pigeonman99, 2024-05-15:
@mwaterbu I've observed the exact same thing on my setup as well.

mwaterbu, 2024-05-15:
I just noticed #124 which seems to be the same as this issue.

hjdhjd, 2024-05-15:
> I have found this as well, but in my case, the incorrect state is only present for at most 5 minutes. What I was finding was that the light would turn on from the motion sensor on the wired wall [...]

My experience is different. It never adjusts itself after a five-minute delay, it's perpetual unless you force a sync, a restart, or toggle the light off/on.

Querying the status too often, I have also found through experimentation, will lead to occasional instability with the Ratgdo device itself. If you're doing occasional automations, it may be fine, but querying every minute or two forever, at least as of the most recent release wasn't a reliable long-term answer.