ratgdo/esphome-ratgdo #268: Question about translating obstruction sensor voltage for Genie and other dry contact type openers.

GingerSnap-xx, 2024-05-01:
In the documentation for Genie type openers, it mentions the operating voltage of the obstruction sensors is too high. Is this something where a level shifter or something could be applied to map the voltage down, but that's outside the scope of the documentation, or is there a reason why even with something to make the voltage "safe" for the hardware, it still wouldn't work?

Thanks!

PaulWieland, 2024-05-01:
Yes you can just use a level shifter or a voltage divider to bring it down to ~6vdc

GingerSnap-xx, 2024-06-11:
Maybe I'm misunderstanding the 2.53i dry contact wiring diagram. I assumed that the obstruction sensors would go to the blue push connector, and then the ratgdo would manage mirroring the inputs through the JST-like connector that goes to the actual GDO . Is it that really we just need to get a reading from the obstruction sensors and don't really need an "in and out" or pass through connection? Said another way, can I just split the connection going from the garage door and send one leg to the obstruction sensors directly and the other leg to the voltage divider so that the voltage the ratgdo sees is lower, and when the obstruction sensors short to ground during obstruction the sensors see ~12vdc drop to ~0 and the ratgdo sees ~3vdc drop to ~0 and both are happy?

GingerSnap-xx, 2024-06-11:
I confirmed that splitting the signal from the GDO and sending one leg to the obstruction sensors and another leg to a voltage divider and on to ratgdo worked just fine. I'll close this out.

closed 2024-06-11 by GingerSnap-xx

ineedhungry, 2025-11-02:
Hey @PaulWieland!  I have a 32 -2.53i, but its a genie garage door using dry contact.  I believe the obstruction voltage is over recommended or at least on the wiring for the 2.5 with dry contacts.