ratgdo/esphome-ratgdo #270: Dry Contact Mode seems to ignore limit switch state

Goose66, 2024-05-07:
Operating in dry contact mode, when you open a door, if you stop it in motion (either from the devices website or the button/keypad) the door continues to report the increasing position over time until finally reporting a state of "OPEN" and a current_operation of "IDLE" even though the open limit switch is never switched "ON". Similarly, closing the door and then stopping the motion causes the ratgdo to report the decreasing position over time and finally "CLOSED" state and "IDLE" current_operation despite the closed limit switch never being switched "ON".

It seems the ESPHome firmware infers the state from the context of the last command and ignores the limit switches. Is this a product of the ratgdo's firmware implementation or a limitation of the ESPHome firmware itself.

null-route, 2025-01-16:
I also noticed this after installing a ratgdo on a Liftmaster LA500 gate using the esphome/dry contact method and integrating it with Home Assistant. Would it be possible to correct this, or is there a recommended workaround?

PaulWieland, 2025-01-16:
See #344

closed 2025-01-16 by PaulWieland