ratgdo/esphome-ratgdo #271: Update ESP Firmware

mclausha1, 2024-05-07:
I am looking to find out how to go about upgrading the firmware I have a V2.5

[image]

I went here to get the bin and tried to update it OTA but it came back as not enough space
https://[paulwieland.github.io/ratgdo/flash.html](https://paulwieland.github.io/ratgdo/flash.html)
which took me here
https://github.com/ratgdo/mqtt-ratgdo/releases and tried to update with the latest file
[image]
 
Once I tried to upload the file to the device it came back and stated not enough space.

How do we update to the latest firmware. I have a issue where mine goes offline and will come back a few seconds later hoping that the update will fix the issue.

kpfleming, 2024-05-08:
If there's not enough flash space to do an OTA update, you'll have to get physical access to the board and update over the USB port.

greggitter, 2024-05-13:
I had same problem and [this](https://github.com/ratgdo/esphome-ratgdo/issues/136#issuecomment-1865903037) resolved it.  Of course you need to verify how much storage your D1 mini or equivalent has.

mwaterbu, 2024-05-17:
No need to get physical access in most cases. The old yaml for this project was only configured for D1 Mini Lite ESP8266 boards, rather than the more common D1 Mini, the difference being that the "Lite" only has 1MB flash, whereas the normal D1 Mini has 4MB. So, as long as you don't actually have the "Lite" you have 4MB of flash memory, but the configuration currently on your board is making it think it only has 1MB to work with, hence the error.

To fix it, you need to get it flashed temporarily with anything that is configured to use the full flash memory, then flash back to your current config (changed to D1 Mini, so it can use all 4MB).

The instructions in the link above should get you going again.

Alternatively, if you change your config to D1 Mini from D1 Mini Lite, and disable the web server as described [here](https://community.home-assistant.io/t/ratgdo-local-mqtt-control-for-your-chamberlain-liftmaster-security-2-0-garage-door-opener/437380/552), the new package should hopefully be small enough to flash. This is a very minimal change to your config, and if you don't need the web server, you're done after one flash. Otherwise, just re-enable it and flash again.