ratgdo/esphome-ratgdo #272: Infinite loop of door opening (power cycling?)

Xaelias, 2024-05-08:
I purchased a ratgdo a few weeks ago and finally got around trying to install it. Except I'm facing an issue that I believe is described also in this [reddit post](https://www.reddit.com/r/homeassistant/comments/1atzkqa/ratgdo_randomly_cycled_my_garage_door_open_and/).

The problem seem to be that when being power cycled (which happens in my case on installation, but also would happen during a power outage for instance), the following things happen.
- The ratgdo boots up
- The control panel boots up and starts "charging"
- When it's done charging, it triggers a door opening, which also seems to restart the panel
- repeat

Any insight?

PS: Security 1+ (purple), with wall mounted panel and obstruction sensors.
Power supply is a 5V@1A usb adapter.

agoode, 2024-05-08:
I suspect this is the same issue as #235. See also #254.

Xaelias, 2024-05-08:
Thanks indeed I missed those. Guess I'll need to compile the firmware locally 😬

Xaelias, 2024-05-09:
So I've tried to disable the emulation mode, I've tried to increase the timeout from 35 to 180s (the panel takes almost 1 minute to come up for me). And the issue is still happening.

If the wall button is not already up and running when the ratgdo comes up, it will start acting erratically and open/close the door randomly. Power cycle it, the problem seem to disappear.

I'm still hoping to find out what's actually happening but as is, I don't know I can leave it deployed :-/

emlynmac, 2024-05-28:
I've worked around this issue by building locally and commenting out the start of the emulation mode in the code.
Specifically, in `secplus1.cpp` around line 55, comment out or remove the `this->wall_panel_emulation();`

To build locally you also need to adjust the includes in the yaml files.

For me that was to change the packages section in `v25iboard_secplusv1.yaml` to point to local packages, rather than the remote urls.
```
 remote_package:
    url: https://github.com/ratgdo/esphome-ratgdo
    files: [base_secplusv1.yaml]
    refresh: 1s
```
needs to be:
```
  esphome-ratgdo: !include base_secplusv1.yaml
```

and in `base_secplusv1.yaml` change:
```
external_components:
  - source:
      type: git
      url: https://github.com/ratgdo/esphome-ratgdo
    refresh: 1s
``` 
to
```
external_components:
  - source:
      type: local
      path: esphome-ratgdo/components
```
making sure the path corresponds to your local git clone.

FireTime, 2024-06-23:
+1 on this. Had a power outage this morning and discovered my garage door opening and closing by itself. 

For those of us with 889LM wall panels we never need the emulation feature. Can emulation be a configurable option in the webUI?

dmzimmerman, 2024-06-23:
+1 from me too. I've had this happen a few times and restarting the ratgdo from its web interface always fixes it, but that's only a viable solution if I'm actually there when it happens...

PaulWieland, 2024-06-24:
The preferred way is to correctly detect if a wall panel is present and aborting emulation mode if any packets are detected.

emlynmac, 2024-06-24:
@PaulWieland, this should work - but with certain panels it takes longer than the code expects and leads to the issues. The simplest fix is allow users who don't need emulation to disable it.

PaulWieland, 2024-06-24:
@emlynmac The current implementation is flawed because it is based on a timer. The correct way to do it is to look for traffic on the data line that did not come from ratgdo itself. If any traffic is detected, abort emulation mode.

emlynmac, 2024-06-24:
> @emlynmac The current implementation in the ESPHome firmware is flawed because it is based on a timer alone. The correct way to do it is to look for traffic on the data line that did not come from ratgdo itself. If any traffic is detected, abort emulation mode.

I agree with this. It sounds like I'll stick with the current local disable code I have for now then.

andy-cooper, 2024-06-24:
Ahhhhhh - thank you @emlynmac - I had been having this problem randomly over the last few months where the garage door opener would suddenly stop working, would open and close randomly, and would require me to reset everything in order to get it working again. I had recently changed the wall panel to an 889LMMC and did not notice this at first - I guess I must have fired it up in exactly the right order. 

I literally just bought a new garage door opener (complete assembly for a total replacement) because I had thought it was the opener itself. For some reason I decided to do one last search for problems and turned up this thread. I've now disabled the emulation mode via using a local clone of the repo as per your guidance. In my installation, I had used the web installation, so I had to make a change to my base yaml as well. Something like the following:

`
packages:
#  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v25iboard_secplusv1.yaml@main
  ratgdo.esphome: !include esphome-ratgdo/v25iboard_secplusv1.yaml
`

For those using Home Assistant, put the cloned git repo in /homeassistant/esphome.

Fortunately the new opener is  sitting unopened on the garage door floor, so I can return it. You just saved me $230!

@PaulWieland - IWBNI you could either advise how to make the fix for checling for traffic on the data line, create a fix for that, or some configuration options to work around the issue.

emlynmac, 2024-06-24:
> Ahhhhhh - thank you @emlynmac - I had been having this problem randomly over the last few months where the garage door opener would suddenly stop working, would open and close randomly, and would [...]

Glad you caught it before it cost you :D

FireTime, 2024-06-24:
> @emlynmac The current implementation in the ESPHome firmware is flawed because it is based on a timer alone. The correct way to do it is to look for traffic on the data line that did not come from ratgdo itself. If any traffic is detected, abort emulation mode.

Is that even possible? Lack of bus arbitration or gatewaying by the ratgdo on the data lines means that unless the keypad happens to send its first message at the correct time the ratgdo and the keypad will step on each other causing corruption and erratic door behavior.  Apart from the current a timeout design (with either a longer or configurable delay)  or a configuration setting either at compile or in eeprom I cant see a robust methodology without some level of data corruption.

PaulWieland, 2024-06-24:
[Sure it's possible](https://github.com/ratgdo/mqtt-ratgdo/blob/a8e213a0643ec40abf0149e2b3ee4daebb2b45d5/src/static_code.cpp#L19). There is plenty of "quiet" time on the line and when the wall panel boots, it repeatedly sends `0x31` for quite a long duration.

Emerica243, 2024-06-30:
@emlynmac @PaulWieland @andy-cooper Forgive my ignorance but I have power blips quite often and it has freaked me out with the door opening\closing infinitely until you reset the ratgdo and ensure it powers up AFTER the wall panel has finally charged back up fully. IT really sucks because its not uncommon for me to get 2-3 blips in a single day sometimes depending on the weather. Point being, what are the options here? I see people are doing some custom compilations of this code and flashing. Way above my knowledge, is that specific code going to be made available to flash? Would getting a security2.0 vs my current security 1.0 GDO solve this? Just looking to solve ASAP.

andy-cooper, 2024-06-30:
@Emerica243 I've just created a fork of this repo with the change needed to remove the emulation for the LM889 wall panel so that we don't get this behavior. Note - it is completely untested, but mirrors exactly what I have done for myself.

To use it, you can do the following:

1. Edit your ratgdo-xxx.yaml file (whatever the name is, it's the one yaml file in your ESPHome HA installation for your garage door).
2. Change the line that reads ```ratgdo.esphome: github://ratgdo/esphome-ratgdo/v25iboard_secplusv1.yaml@main``` and instead make it read ```ratgdo.esphome: github://andy-cooper/esphome-ratgdo/v25iboard_secplusv1.yaml@main```
3. Save this file
4. Click "Install" on the right hand side of the screen, select "Wirelessly" or however you normally install it.
5. Wait for things to be installed.

To test it once the software update and install has completed, pull the power on your garage door opener for a few minutes. Then power it up. You should notice the LM889 power on, but not be ready immediately. Once it's ready, there should be 3 beeps, and that's it. No random opening/closing of the garage door.

LMk how it goes.

Emerica243, 2024-06-30:
> @Emerica243 I've just created a fork of this repo with the change needed to remove the emulation for the LM889 wall panel so that we don't get this behavior. Note - it is completely untested, but [...]

Thank you very much for taking the time to explain this to me! I only have experience with using the ratgdo ESPHome web installer. Which gibes no selection or granularity on whats being flashed to the board. I poked around my HA instance and see no evidence of any YAML files. Im using the ESPHome integration which just brings the ratgdo device into HA because of its ESPHome presence. 

I see on git-hub the correlating .yaml file youre mentioning for my device ( https://github.com/RATGDO/esphome-ratgdo/blob/main/static/v25iboard_secplusv1.yaml ) 

I also see the line youre referencing inside that yaml file( dashboard_import:
  package_import_url: github://ratgdo/esphome-ratgdo/v25iboard_secplusv1.yaml@main ) 

But im a bit lost on how im doing any updates to the board after modifying anything. Theres nothing regarding ratgo yaml files or performing any sort of flashing\updating of stuff like this within my HA or on the ratgdo esphome web interface.

andy-cooper, 2024-07-01:
@Emerica243 - sorry for the slow response. Yes, you want to have the ESPHome HA add-on installed. Once it is installed, you may need to "adopt" the ratgdo device. 

Once installed and adopted, you'll see something like the following:

[image]

If you click on the "edit", you'll get the YAML

andy-cooper, 2024-07-01:
@Emerica243 - to "adopt" the device, you go to "Settings > Devices & Services" > ESPHome, and then click on "Add". It will ask you to add the host and port settings for the ESPHome device, in this case your garage door opener. Once you've done that and successfully added it, you should be able to edit it as described.

Emerica243, 2024-07-01:
> @Emerica243 - to "adopt" the device, you go to "Settings > Devices & Services" > ESPHome, and then click on "Add". It will ask you to add the host and port settings for the ESPHome device, in this [...]

Got it all done! Thanks again Andy!

andy-cooper, 2024-07-01:
Fantastic! I hope this solves the problem for you. Report back here when you've tested it with a power outage.

PaulWieland, 2024-07-12:
Please update to latest release

closed 2024-07-12 by PaulWieland