ratgdo/esphome-ratgdo #274: Up vs Down Arrow In Home Assistant

rtvanhook, 2024-05-10:
Brilliant piece of work.  Works flawlessly.  I hate to bring up the 0.01% when 99.99% is amazing, but...

The up and down arrow icons seem to be exchanged in the HA UI.  I'm integrating via ESPHome firmware.

See attached screen shot.

[image]

In the HA device screen for "Door", it shows an up arrow, a square, and a down arrow.  At least that's how the icons strike me.

Best I can tell, either the up or down arrow is solid and the other grayed to show the door open or closed.

To me, a solid up arrow means open (i.e., door up), and a solid down arrow means closed (i.e., door down).

That doesn't seem to be the case.  That screen shot is with the door closed, and the up arrow is solid.

Maybe it's just my brain.  Again...it works fine, and thank you so much for sharing your brilliance with the world.

PaulWieland, 2024-05-10:
The icons are buttons for controlling the door. When the door is closed, you can't close it and vice versa.

closed 2024-05-10 by PaulWieland

rtvanhook, 2024-05-10:
So the up arrow is telling me?  Intuitively, it tells me that the door is up.  I'm certainly willing to recalibrate.

rtvanhook, 2024-05-10:
Problem with intuition is that everyone's works differently.  Sorry.