ratgdo/esphome-ratgdo #277: Dubious ownership.  How can I fix this?

FutureTense, 2024-05-16:
> INFO ESPHome 2024.5.0
> INFO Reading configuration esphome/ratgdov2-284fdb.yaml...
> INFO Updating https://github.com/ratgdo/esphome-ratgdo@None
> Failed config
> 
> external_components: [source esphome/.esphome/packages/74f605c3/base.yaml:4]
>   - 
>     fatal: detected dubious ownership in repository at '/config/esphome/.esphome/external_components/74f605c3'
>     To add an exception for this directory, call:
>     
>     	git config --global --add safe.directory /config/esphome/.esphome/external_components/74f605c3.
>     source: 
>       type: git
>       url: https://github.com/ratgdo/esphome-ratgdo
>     refresh: 1s
> 

I'm getting the message and I'm not sure why.  What are my options for getting my ratdgo working nicely with esphome and git again?  Preferably one that doesnt have me climbing up a ladder to get the ratgdo box.  Isn't there some **sudo** goodness that can make git happy agin?

restauffer, 2024-05-16:
I am not an expert but it seems your YAML is pointing to a wrong Github repository. Can you send a copy of your YAML file so we can look at it?

If you don't want to send the file, you can try this. You may be able to connect via OTA or you may need to connect with a USB cable to your computer:

1) Make a copy of the current YAML file in case you need/want to revert due to the following not working.
2) Go to https://github.com/ratgdo/esphome-ratgdo/tree/main/components/ratgdo and find the YAML file that matches your board. If you are not sure what board you have, there is a corresponding png file with an image of each board.
3) Copy and keys and passwords from the file you saved (current file) to this new one.
4 Copy the new YAML file into the ESPHOME profile  for your RATGDO and install it.

rlowens, 2024-05-16:
> but it seems your YAML is pointing to a wrong Github repository.

No, this is just a ownership issue of the local folders. They need to re-setup ESPHome or manually correct the folder ownership issue.

How do you have ESPHome installed?