ratgdo/esphome-ratgdo #278: Mystery Open Button in Web UI Lock Remotes Section

mwaterbu, 2024-05-16:
What does the third button do?
[image]

The [documentation](https://ratgdo.github.io/esphome-ratgdo/webui_documentation.html) also shows this button, but does not mention what it is supposed to do.

Clicking the button gives this log message:
`'Lock remotes' Does not support Open.`

restauffer, 2024-05-16:
The rightmost column of the documentation gives a decent explanation:

Lockout the wireless remotes.
🔐 - lock the remotes. Wireless remotes will be ignored by the GDO.
🔓 - unlock the remotes. The GDO will respond to commands from wireless remote

This feature allows you to disable the use of remotes for whatever reason (lost remote, going out of town, etc.) If I remember correctly, it does not disable the wireless keypads.

rlowens, 2024-05-16:
> the documentation gives a decent explanation:

It does NOT mention this non-working "Up Arrow" button which this issue is about.

> If I remember correctly, it does not disable the wireless keypads.

That would be strange to me since I thought wireless keypads are just remotes, but I don't own one to test.

mwaterbu, 2024-05-17:
> It does NOT mention this non-working "Up Arrow" button which this issue is about.

Yes, this is what I am asking about, not the lock and unlock buttons.

> > If I remember correctly, it does not disable the wireless keypads.
> 
> That would be strange to me since I thought wireless keypads are just remotes, but I don't own one to test.

Definitely strange, but it does seem to be the correct intended behavior, according to [this](https://www.youtube.com/watch?v=pEIlktVI1Wk&t=41). I was questioning myself until I found that video, as I thought the lock wasn't functioning. I thought I remembered it locking out all of the controls in the past, not just remotes, but my memory must be faulty.

jacobpoder, 2024-05-17:
> Definitely strange, but it does seem to be the correct intended behavior, according to [this](https://www.youtube.com/watch?v=pEIlktVI1Wk&t=41). I was questioning myself until I found that video, as [...]

On mine it locks out my two remotes, but not the wall-mounted button (even though that is also wireless)

briodan, 2024-05-18:
the up arrow is the lock open action as defined in the esphome lock domain

https://esphome.io/components/lock/index.html#lock-open-action.

rlowens, 2024-05-18:
How can it be disabled from the Web UI since

> 'Lock remotes' Does not support Open.