ratgdo/esphome-ratgdo #279: Keypads not locked out

briodan, 2024-05-18:
I just got my ratgdo v2.53 running it with a chamberlain security 1+ door opener.

Noticed that when using the lock remotes function the "regular" remotes like you have in your car get locked out but not the wireless keypad.

Not sure if this is intended behaviour or an issue, but seems strange that one kind of wireless remote gets locked out and another does not.

FrankJaco, 2024-05-20:
That apparently is the intended behaviour per Chamberlain. The Ratgdo is just tapping into their logic. This is directly from Chamberlain.....
_Lock button
This feature is designed to prevent operation of the door from hand-held remote controls. However, the door will still open and close from the Push bar and the Keyless Entry. To activate: Press and hold the Lock button for 2 seconds. The Push bar indicator light will flash as long as the Lock is on....._

I too would like the ability to lock out even the keypad on some occasions for added security.  There might be a way to do that with ESPhome, or perhaps via home assistant and a smart plug. Have the plug cut power to the entire gdo at night or when away on vacation. (If you have a battery backup on your gdo like I do, beyond the smart plug an inline smart relay to disconnect the battery would be needed too.   I wonder if killing the power and disconnecting the battery nightly would do any harm to the gdo?   Anyone have any thoughts on this?

fj

fj

jerrm, 2024-05-20:
May not be what you want, but many (all?) third party keypads pair as a remote and won't override the lock command.

FrankJaco, 2024-05-20:
I didn't realize that.  That might very well be a simpler solution. Perhaps less "fun" to implement, however simpler is usually better.
Can you point me to a particular 3rd-party keypad that for sure acts like a remote that that personally like?

Thx
fj

jerrm, 2024-05-21:
The one I returned was https://a.co/d/bTX1w5g. 

You give up the no-pin close functionality.