ratgdo/esphome-ratgdo #280: Obstruction Status Always "On" in HomeKit

franktastical, 2024-05-21:
Description

I am experiencing an issue where the "Obstruction" status is always "On" in HomeKit, causing it to report an issue even when there is nothing blocking the door. Despite this, the door operates normally with remotes, wall button, HomeKit, or the ratgdo web UI. When the sensor is actually obstructed, it functions correctly in all modes (won’t close).
**Environment**

- **Firmware Version**: 2024.4.1 (Apr 29 2024, 04:18:43)
- **Device Model**: Chamberlain B2405T
- **Voltage Readings**:
  - Unblocked: 6V
  - Blocked: 6.2V

**Steps to Reproduce**

1.  Observe the "Obstruction" status in HomeKit.
2. Ensure there is no physical obstruction.
3. Check the operation of the door using remotes, wall button, HomeKit, and the ratgdo web UI.

**Observations**

- The "Obstruction" status remains "On" in HomeKit despite no physical obstruction.
- Door operation is unaffected in all control modes.
- Voltage readings from the white sensor port on the opener to the black sensor port on ratgdo are 6V (unblocked) and 6.2V (blocked).

**Additional Information**

Please let me know if any further information or diagnostic steps are needed.

Thank you!

PaulWieland, 2024-05-21:
And what is the obstruction status showing in ESP Home?

rlowens, 2024-05-21:
Also, how are you connecting the ESPHome firmware to HomeKit? (There are a few different ways to do it.)

franktastical, 2024-05-21:
Sorry, I meant to include those details in the original post.  Not sure how I omitted them.

I have ratgdo connected to HomeKit via homebridge-ratgdo plugin on a child bridge.  The web UI for the ratgdo shows "Obstruction" as "On" which is then passed along to HomeKit where I see it.  Triggering the sensor doesn't seem to cause any output on the ratgdo UI, but does cause the light on the sensor to flash.

[image]

franktastical, 2024-05-21:
And I have thrice verified the wiring is correct and nothing is shorted out.  Multi metered all the connections to be sure.

frankea, 2024-06-20:
I've been messing around with various wiring setups and still no luck.  Any ideas on the software side of things?

mac-tron, 2024-12-17:
From my testing, in my set up this only happens when I introduce HomeKit to my set up.

ratgdo-homekit firmware, obstruction detected and I cannot close the door.
ratgdo-esphome firmware, this works better with no obstructions until I connect HASS to homebridge. When I delete homebridge and reboot the system the obstruction is gone.

So for whatever reason as soon as Homekit is introduced (even with the same wiring) it flags an obstruction.