ratgdo/esphome-ratgdo #282: 2.53i online and responding - but door isn't moving.

DaveBKC, 2024-05-28:
Hello everyone. I've hit a roadblock and would appreciate any thoughts. Here's my setup:

- Ratgdo Board: 2.53i flashed with ESPHome version 2024.5.3
- Opener: Chamberlain RJO70 (side mount Security 2.0)
- Opener button and object sensors are wired into the blue terminal connectors.
- The r/b/w three-wire connector is wired into the opener.
- WiFi coverage in garage is good.

All the pieces seem to be in place, but it's not controlling the opener. You can see the message below: "The door is not moving." 
I've got some unrelated issues with my Home Assistant integration, so for now I'm trying to get the Ratgdo to work by controlling it directly. 

Can someone tell me if these incomplete packets are normal?
Any thoughts on using the current version of ESPHome, an older version, or maybe ditch it for MQTT?

```
17:24:22	[W]	[ratgdo_secplus1:262]	[17355376] Discard incomplete packet: [39 ...]
17:24:23	[W]	[ratgdo_secplus1:262]	[17355621] Discard incomplete packet: [38 ...]
17:24:23	[W]	[ratgdo_secplus1:262]	[17355896] Discard incomplete packet: [3A ...]
17:24:23	[W]	[ratgdo_secplus1:262]	[17356140] Discard incomplete packet: [39 ...]
17:24:23	[W]	[ratgdo_secplus1:262]	[17356399] Discard incomplete packet: [38 ...]
17:24:24	[W]	[ratgdo_secplus1:262]	[17356659] Discard incomplete packet: [3A ...]

...

17:26:28	[W]	[ratgdo_secplus1:262]	[17480591] Discard incomplete packet: [3A ...]
17:26:44	[D]	[button:010]	'Toggle door' Pressed.
17:26:44	[W]	[ratgdo_secplus1:262]	[17497289] Discard incomplete packet: [38 ...]
17:26:54	[D]	[cover:076]	'Door' - Setting
17:26:54	[D]	[cover:080]	  Command: STOP
17:26:54	[W]	[ratgdo:490]	The door is not moving.
17:26:58	[D]	[cover:076]	'Door' - Setting
17:26:58	[D]	[cover:084]	  Position: 100%
17:27:08	[D]	[cover:076]	'Door' - Setting
17:27:08	[D]	[cover:080]	  Command: STOP
17:27:08	[W]	[ratgdo:490]	The door is not moving.
17:27:08	[W]	[ratgdo_secplus1:262]	[17520772] Discard incomplete packet: [38 ...]
17:27:08	[W]	[ratgdo_secplus1:262]	[17521045] Discard incomplete packet: [3A ...]
17:27:08	[W]	[ratgdo_secplus1:262]	[17521289] Discard incomplete packet: [3A ...]
17:27:08	[W]	[ratgdo_secplus1:262]	[17521564] Discard incomplete packet: [3A ...]
17:27:09	[W]	[ratgdo_secplus1:262]	[17521822] Discard incomplete packet: [39 ...]
17:27:09	[W]	[ratgdo_secplus1:262]	[17522070] Discard incomplete packet: [38 ...]
17:27:09	[W]	[ratgdo_secplus1:262]	[17522342] Discard incomplete packet: [3A ...]
17:27:10	[W]	[ratgdo_secplus1:262]	[17522601] Discard incomplete packet: [38 ...]
17:27:10	[W]	[ratgdo_secplus1:262]	[17522859] Discard incomplete packet: [3A ...]
```

bdraco, 2024-05-28:
You have the security+ 1.0 code installed `ratgdo_secplus1`

DaveBKC, 2024-05-28:
Oh brother. Thank you. It's not letting me do a wireless update at the moment but I'll test try that asap.

closed 2024-05-28 by DaveBKC