ratgdo/esphome-ratgdo #283: Feature Request:  GDO backup battery health and percentage

FrankJaco, 2024-05-31:
I have finally migrated to the ESPhome firmware from MQTT with my Sec1.0+, purple learn button Chamberlain HD9100.  Alas, my wall panel is digital, thus I needed to eliminate it to make the gdo work with Ratgdo (or replace it with an analog panel). I have recreated most of the features that I need directly in Home Assistant using a smart button and PIR sensor to replace the wall panel's common functions.  One feature that I sorely miss that I do not believe is possible by purchasing an analog wall panel has to do with my battery backup. The digital wall panel would provide info on the back up battery's life and informed me when it needed to be replaced. 

I would like to see a feature added to the ESP version of Ratgdo to extract that info from the gdo. The info has to be in the gdo's logic itself.   I am aware there are ways another ESP8266 could be fitted with a sensor (or perhaps the Ratgdo itself), but I don't think that is necessary and it is something that I cannot take on at this time.  The data must be in the gdo, as my digital wall panel has no such sensor in it.

Thoughts?  

fj

FrankJaco, 2024-06-02:
A little more info.....   The logic for this is for sure in the GDO itself, as there is a LED on the GDO that indicates the battery condition. The smart wall panel will receive and display one of the following status from the GDO':

BATTERY BACKUP ENABLED 
(Meaning: Opener is running on battery power with full charge.)

BATTERY LOW BATTERY
(Meaning: Opener is running on battery power with low charge.)

BATTERY BAD REPLACE BATTERY NOW SEE OWNER’S MANUAL
(Meaning: Battery is fubar.)

POWER RESTORED BATTERY CHARGING
(Meaning: Battery charging from recent battery backup state.)

Would it be possible somehow to monitor for those same 4 messages in a similar manner that the ratgdo software determines the various door, light and lock states?  At least on Sec 1.0+ the ratgdo logs constantly displays these states nonstop....    

Time	level	   Tag	          Message
10:03:30	[D]	   [ratgdo:215]	  Lock state=UNLOCKED
10:03:30	[D]	   [ratgdo:100]	  Door state=CLOSED
10:03:31	[D]	   [ratgdo:209]	  Light state=OFF
10:03:31	[D]	   [ratgdo:215]	  Lock state=UNLOCKED
...

I am not sure what exactly the "Tag" code is but am guessing it is the code that represents the message?  If so, could it be possible to determine the code of the 4 battery states and pass on the state as an entity or an attribute?

fj

timtIAD, 2025-06-02:
It would be great to see these features added. Thanks for writing this up.

NukeThemTillTheyGlow, 2026-08-18:
Add my 2 cents, I recently had a 3 day power outage while out of town.  Came home expecting 2 of my 3 garage doors that have battery backup (all 3 using RATGDO's), but only one worked (after starting up network/wifi).  Reason, dead battery.  then the next day after I think 2 cycles, the one that worked died.    Not the end of the world, but I'd love to set up an automation in HA OS to track the status and alert me ahead of time.