ratgdo/esphome-ratgdo #288: Unable to communicate with GDO

davidm2232, 2024-06-17:
Saturday my ratgdo stopped working after months of perfect operation. The LED on the wall control is dead and I cannot close the door from the wall button or through Home Assistant. I was initially able to open and close the door with the remote but now the lights on the opener just flash. I initially plugged in a USB to the ratgdo and got logs stating 'not connected to GDO'. I reflashed the device several times without any improvement. I followed this guide (https://github.com/ratgdo/esphome-ratgdo/issues/136) to try and reflash OTA and was able to do so successfully but the opener still won't respond to any inputs. Logs attached. Any help would be appreciated. 
`
[logs_ratgdov25i-4b0111_logs.txt](https://github.com/user-attachments/files/15876171/logs_ratgdov25i-4b0111_logs.txt)
`

davidm2232, 2024-07-01:
I was able to fix this by replacing the wall control button.

PaulWieland, 2024-07-02:
Another wall control panel bites the dust

closed 2024-07-02 by PaulWieland