ratgdo/esphome-ratgdo #289: ESPHome 2024.6.0 fails validation: At least one platform must be specified for 'ota'; add 'platform: esphome' for original OTA functionality

flavio-fernandes, 2024-06-19:
```
INFO ESPHome 2024.6.0
INFO Reading configuration /config/esphome/ratgdov25i-e51133.yaml...
INFO Updating https://github.com/ratgdo/esphome-ratgdo@None
INFO Updating https://github.com/ratgdo/esphome-ratgdo@None
INFO Detected timezone 'America/New_York'
Failed config

At least one platform must be specified for 'ota'; add 'platform: esphome' for original OTA functionality
```

[image]

xdaben, 2024-06-19:
You can add

```
ota: 
  platform: esphome
```

to your yaml for the device to get it working again

PaulWieland, 2024-06-20:
Yes it should be part of the base yamls

PR #293 (2024-06-23): OTA: Moving to base.yamls and adding ESPHome platform
Original PR https://github.com/ratgdo/esphome-ratgdo/pull/291

Moving the OTA block to the base of the respective
esphome-ratgdo board.

Starting with version 2024.6.0, at least one platform must
be specified for 'OTA'; add 'platform: esphome' for
original OTA functionality.

Fixes: #289
closed 2024-06-23 by bdraco