ratgdo/esphome-ratgdo #290: ESPHome 2024.6.0 mitsubishi-heatpump KeyError: 'UART0'

korttoma, 2024-06-20:
INFO ESPHome 2024.6.0
INFO Reading configuration /config/esphome/mitsubishi-heatpump.yaml...
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1079, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1066, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 484, in command_run
    exit_code = write_cpp(config)
                ^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 193, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 205, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 681, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 246, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 185, in wrapped
    await coro(conf)
  File "/esphome/esphome/coroutine.py", line 80, in coro
    ret = yield from _flatten_generator(gen)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/coroutine.py", line 118, in _flatten_generator
    val = gen.send(to_send)
          ^^^^^^^^^^^^^^^^^
  File "/data/external_components/3b4567cc/components/mitsubishi_heatpump/climate.py", line 66, in to_code
    serial = HARDWARE_UART_TO_SERIAL[config[CONF_HARDWARE_UART]]
             ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyError: 'UART0'

korttoma, 2024-06-20:
Found a solution here -> https://github.com/geoffdavis/esphome-mitsubishiheatpump/issues/152

sanjay900, 2024-06-20:
Not sure why you opened this here being that its a totally different integration haha

korttoma, 2024-06-20:
> Not sure why you opened this here being that its a totally different integration haha

Yeah, sorry. i updated ESPHome so I thought that the issue was there.

closed 2024-06-20 by korttoma