ratgdo/esphome-ratgdo #294: ratgdo 2.52i - Obstruction Sensor Works, But Can't Control Lights or Door

davertron, 2024-06-23:
Hi folks,

Quick Details:

Hardware: ratgdo 2.52i
Firmware: 2.56 (Current), Also tried 2.57
GDO: **Overhead Door Odyssey 1000**
GDO Control Protocol: Sec2 and Sec1

Overall my issue seems very similar to #122, but the solution mentioned in that issue (using terminal 0 instead of 1) doesn't really translate to my situation since I have a different model gdo and pin layout.

I can open my garage door via the wall switch fine. Before I swapped the wiring (shown in the picture below), my obstruction sensors didn't light up and I couldn't use the wall switch to open/close.

Now, at some point, I thought that I read somewhere that Overhead door openers were just rebranded chamberlain or liftmaster units, but I can't actually find a source for that anymore so I may have dreamt it up. So it's possible that ratgdo will just not work on my door opener. However, it does seem to be able to get obstruction status fine, as I can see it updating in home assistant properly. Still, perhaps there are slight differences in the control protocol that overhead door has made to make it incompatible or non-standard, I obviously don't know enough about all of this to say. I do suspect there might be slight differences since they obviously want to sell their own "smart panel" to control your opener and make you pay a monthly fee and all of that.

I can say that I had to wire it up slightly differently than I was expecting based on the wiring diagram. Here's a picture of how my gdo was wired up before I installed the ratgdo (top) and after (bottom):

[image]

And this is a picture of the wiring on the ratgdo (note: Garage door wires were bigger so I spliced smaller jumpers to make it easier which is why they aren't the big white wires shown in the picture above):

[image]

I've attached logs I took while telling the door to do various things, like open and close, turn on / off the light, etc. It seems like it was able to get rolling codes and it thinks it's writing them successfully, but the door doesn't actually do anything. I've tried with both sec2 and sec1 control protocol, which is why the two logs.

The thread mentioned above also mentions resetting your client id in order to get new rolling codes, but I don't see where to do that in the web interface.

I also tried with firmware 2.57 flashed on the ratgdo, but same results.

I've also attached the owners manual for my gdo, just in case there's anything useful in there.

Appreciate any tips anyone might have on things I could try!

[esp-web-tools-logs (sec1).txt](https://github.com/user-attachments/files/15945131/esp-web-tools-logs.sec1.txt)
[esp-web-tools-logs (sec2).txt](https://github.com/user-attachments/files/15945132/esp-web-tools-logs.sec2.txt)
[Odyssey Destiny Owners Manual.pdf](https://github.com/user-attachments/files/15945134/Odyssey.Destiny.Owners.Manual.pdf)

PaulWieland, 2024-07-12:
Try the dry contact control protocol.

apetti1920, 2024-07-24:
I am having the same issue.  This door opener does not have the required ports available for the dry contact method.

ckuykendall81, 2024-09-11:
The only thing I can get to operate this gdo is the button press, the cover stuff doesn't work at all.  Button doesn't show in HomeKit properly.  Flashed to the HomeKit and nothing works there either

PaulWieland, 2024-09-12:
The HomeKit firmware does [not yet support dry contact control](https://github.com/ratgdo/homekit-ratgdo/issues/32). 

Limit switches are required in order to have status feedback as well as discrete controls. Without them the only thing that will work is the toggle control.

the-gray-ghost, 2024-12-07:
I am trying to get this opener to work and found some interesting things.
- ratgdo v2.53i flashed with the Security + 2.0 protocol
- Wired up like jimschley suggested.

1. If the door is open when I plug the ratgdo into power, the door closes half way, stops, then about 15 seconds later opens fully.
2. If the door is closed when I plug it in, about 30 seconds later it opens fully. (I cannot tell if it attempts the close/stop cycle)
3. The door does not appear to respond to HA commands (open/close/toggle).
4. The open/shut status is not updated in HA, however the obstruction sensors provide real time updates.

PaulWieland, 2024-12-14:
This is a dry contact control opener. Dry contact control was just recently added to the HomeKit firmware.

closed 2024-12-14 by PaulWieland

danberger, 2025-06-15:
Thank you @jimschley for the solution! That made it so easy. Were you able to get the state of the door returned?

jimschley, 2025-06-15:
> Thank you [@jimschley](https://github.com/jimschley) for the solution! That made it so easy. Were you able to get the state of the door returned?

I've only been able to get the "toogle door" function to work