ratgdo/esphome-ratgdo #309: Dry Contact - Discard Incomplete Packet

kramttocs, 2024-07-14:
Finally got around to flashing my 2.5i for esphome dry Contact. It seems to work for a bit (status, switches, open/close action) but then stops.
The log is filled with the Discard Incomplete Packet message. It references secplus1. Does the dry Contact version build on secplus1? The GitHub code looks like it does so maybe seeing this is expected and not related to why I am getting the error.

The unit is in a shop building but there is a repeater inside near it so I don't believe it's a poor signal issue.

kramttocs, 2024-07-14:
Just had a thought - I wonder if the online flasher correctly flashed the dry Contact version but perhaps when I renamed the name/hostname it pulled the secplus1 down by mistake?

kramttocs, 2024-07-14:
When viewing the config, it did have the secplus1 yaml link in there. I updated it to be dry Contact, saved, and installed.
It's back to looking like it should work the correct status and all. Will keep an eye on it and see if it changes back to secplus1 after a future esphome update.

PaulWieland, 2024-07-14:
If it says secplus1 then you installed the wrong firmware. Choose "Dry contact control" from the firmware installation page.

[image]

kramttocs, 2024-07-14:
That's the thing - I did it twice and selected that both times.

kramttocs, 2024-07-14:
I do t know if this comes into play but line 28 here is what I was referring to earlier: https://github.com/ratgdo/esphome-ratgdo/blob/main/static/v25iboard_drycontact.yaml

Should it be referencing secplus1?

PaulWieland, 2024-07-14:
Yes, you are right. I updated the path, please try again and let me know if it fixed it.

closed 2024-07-15 by PaulWieland