ratgdo/esphome-ratgdo #311: Sec 1.0 discarded packets even on latest release, device still entering emulation mode, still causing door to open and close

jgluck, 2024-07-22:
Sec 1.0 discarded packets even on latest release, device still entering emulation mode, still causing door to open and close. 

For some reason, occasionally my 889LM takes about 30 seconds to start up. Is there any way to extend the wait before entering emulation mode to 1 minute? Or make it configurable?

[image]

Even now, our door is opening and closing on its own.

jgluck, 2024-07-22:
Short term I have flashed andy-cooper’s branch from #272 , and we’ll see if it sticks for now. I’m happy to pull any logs necessary. I’m entirely willing to believe that I have a faulty 889LM; however, even so I’d prefer the negative outcome to be the door opener doesn’t respond, instead of randomly opening and closing.

PaulWieland, 2024-07-23:
I would be interested in seeing the serial log if you can share it. From as close to startup as you can get it.

jgluck, 2024-07-23:
I’ll try to grab one. @PaulWieland are those just the streaming logs available via the ESPHome plugin (or on the ESPHome landing page?)

PaulWieland, 2024-07-23:
From the esphome webui on the device itself.

gullygossner, 2024-08-19:
are there any updates on this issue?

PaulWieland, 2024-08-19:
Still waiting on a serial log showing it happening.