ratgdo/esphome-ratgdo #313: Liftmaster LM70 EVF does not connect to ratgdo, random small open and close at startup and sync, but no status or control

SiebeHodzelmans, 2024-07-30:
Hello,

I received a ratgdo v 2.53i board last week, and tried to install it this weekend. Unfortunately, it doesn't seem to connect properly to my garage door opener (Liftmaster LM70evf). I flashed esphome-ratgdo, with security 2.0.
When I connect the ratgdo, it opens the door a small amount (motor runs 2 or 3 seconds), then it pauses, then the door closes again for two or three seconds. Sometimes this open and closing also repeats one or two times. 
On the ESPhome webpage, no status is reported, and query buttons do nothing. The buttons to close or open the garage door also do nothing. When I press the sync button, the same random open and close happens again. When I close or open the garage with my remote or the buttons on the back of the garage opener, no change is reported in the webUI or the logs.

Troubleshooting I already tried:
- Reboot ratgdo with it being connected to the garage door opener
- Reboot garage door opener, with ratgdo already powered on
- Double checking my wiring (see also picture attached)
- Reflash (using the web flash method), with erasing the flash

My question is, what can I do next for troubleshooting, or what am I missing here?

Gist with the logs: https://gist.github.com/SiebeHodzelmans/f893cd56c71f8657dbcf3c60e139fb7c

[image]

torandreroland, 2024-07-31:
You have a RestOfWorld garage door opener (see https://github.com/PaulWieland/ratgdo/wiki). You need to plug the red wire into port 0, not 1.

SiebeHodzelmans, 2024-08-02:
Thank you so much. That was indeed my issue. As a non-native English speaker I wasn't familiar with the abbreviation ROW.

closed 2024-08-02 by SiebeHodzelmans