ratgdo/esphome-ratgdo #315: Debounce tweak- REQUEST FOR ACCESS

thomasshawn, 2024-08-01:
Have 5 units. one of them has an inconsistent ability to see OPEN state. I have tried multiple dry contact sensors and tweaked their position to the point I have determined I just need a slightly longer debounce time. I see where it is configured in the base drycontact yaml but that is an external component that we don't have easy access to make changes.

**can we get an option on the devices config URL to dial in this number for the dry contact**  that would be the easiest end-user modification.

or Is there a way to add a line to the device yaml to overrride that number?

if not, would it be reasonable to clone to repo and then make the change on my own version and redirect my unit to reference that repo rather than the offical?

PaulWieland, 2024-08-01:
This is something I would like to implement, but doing so is going to be difficult because I don't know if you can modify a filter value on the fly in esphome. If not, that would mean implementing a custom GPIO output with custom debounce logic.

For now you can just clone the repo and build locally with your own debounce time.

thomasshawn, 2024-08-01:
Thank you. I figured that might be the most sensible but I didn't want to do all of that only to find out you were soon to be releasing or already had implemented in some way. 
Thanks again for your great product

PaulWieland, 2024-08-01:
Actually, you really only need to copy the YAML files. They are pointed at the git repo and will pull the src down automatically.

thomasshawn, 2024-08-01:
I'll give that a try, probably this weekend. Thanks.