ratgdo/esphome-ratgdo #316: Where do I find bin files for ota updates

bryan-d-garcia, 2024-08-03:
I have several Ratgdo devices installed and I am trying to do an OTA update, but I cannot find the needed bind files.

My Chamberlain is using Security + 1.0, I am using a ratgdo v2.5x board.   I am using ESPHome with the Crestron Home Chowmain drivers.

How do I find the necessary Bin file so I can use the OTA feature to update to the latest firmware?

Thank you
Bryan

restauffer, 2024-08-04:
Try https://ratgdo.github.io/esphome-ratgdo/

You select your opener type and then your board.

Ricke

bryan-d-garcia, 2024-08-04:
Thank you.  I am aware of this, but it does not provide me the bin files so I can do an OTA update!BryanSent from my iPhone

restauffer, 2024-08-04:
Sorry. It’s been awhile since I used that interface and I forgot that only works with a wired connection.

I use the HA integration but even with the esphome web interface I am pretty sure you don’t need a bin file. You need a yaml file and from that a bin file is created when it complies. The yaml files are here:

https://github.com/ratgdo/esphome-ratgdo?tab=readme-ov-file

scroll down to find the list

I hope I’m not giving more wrong advice.

Ricke

jcarr, 2024-08-05:
- Find your IP of the ratgdo.  
- Identify your board type.  Mine is a 2.5i and I'm using Security2.0.  For my board, it will be the v25iboard.yaml manifest.
- Make sure you install the `esphome` utility (`brew install esphome` on a mac)
- Clone the repo:
`git clone https://github.com/ratgdo/esphome-ratgdo/`
- cd into the directory
`cd esphome-ratgdo`
- Run `esptool`.  This will compile and and upload your new firmware:
`esphome run v25iboard.yaml --device <your_ip>`

I checked and mine was updated to 2024.7.3.

bryan-d-garcia, 2024-08-05:
Thank you.   I will give this a try.   Need to figure out how to install on
my Windows PC.   Thank you for the help.

jcarr, 2024-08-05:
There's instructions on how to install esphome on the esphome site: https://esphome.io/guides/installing_esphome.html#windows

I don't know if git is included but you can always download the zip file, it should be the same contents.  You might need PlatformIO and vscode to build it.

C4Wiz, 2024-10-22:
https://github.com/ratgdo/esphome-ratgdo/actions

click the most recent build, scroll to bottom and click on your board, a zip file will be downloaded, extract zip, open folder and the ota.bin file is inside