ratgdo/esphome-ratgdo #317: password protect ESPHome web interface

ShaneBoland, 2024-08-04:
This board is great I purchased 2 and plan to order more for family members. However, I'm concerned that the interface doesn't require credentials. How do you password protect access to the ESPHome web interface? Although I plan to isolate the ratgdo with firewall rules, I want to also password protect access to both the web interface as well as requiring credentials for home assistant on the initial initial connection. Any help would be much appreciated. Thank you

restauffer, 2024-08-04:
See https://esphome.io/components/web_server.html#configuration-variables 

for info on how to put credentials on the esphome web server component.

See 
https://www.home-assistant.io/docs/authentication/

for info on setting credentials for HA. A vpn is advisable if you want access from outside your home network.

Ricke

billmoseley, 2024-08-19:
Is there a problem simply not running the web server? The docs do say: 

> Please note that enabling this component will take up a lot of memory and may decrease stability, especially on ESP8266.

Maybe disabling solves the security issue and improves stability.

You can back out `web_server:` if using `packages:` like this:

```
packages:
  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v25iboard.yaml@main

web_server: !remove
```

closed 2024-09-28 by PaulWieland

SnapEscalatorCarrousel, 2025-08-22:
Maybe I'm dumb, but the only way I was able to edit the yaml config for my ratgdo32 disco was using ESP Home Builder which I installed on my Raspberry Pi 4b running 64 bit linux (NOT home assistant OS) using:

`sudo -s
apt-get install pip -y
apt-get install pipx -y
apt-get install python3-tornado -y
pipx install esptool
pipx ensurepath
pipx completions

Log out and log in

docker pull [ghcr.io/esphome/esphome](http://ghcr.io/esphome/esphome)
docker run --rm --net=host -v "${PWD}":/config -it [ghcr.io/esphome/esphome](http://ghcr.io/esphome/esphome)
http://raspberrypi.local:6052/`

I'm not actually sure this is all necessary.

I adapted from following sources:
1. https://esphome.io/guides/getting_started_command_line/#esphome-device-builder-docker
2. https://stackoverflow.com/questions/75608323/how-do-i-solve-error-externally-managed-environment-every-time-i-use-pip-3