ratgdo/esphome-ratgdo #318: Dry contact toggle does not toggle door

cbuckley, 2024-08-05:
Using the new open/close dry contacts toggle feature merged in https://github.com/ratgdo/esphome-ratgdo/pull/307 I can see that the two contacts are marked as `ON` in the UI and the toggle debug message is displayed but the door never moves and the position stays at 0%

This works sometimes, usually after a reboot but not after every reboot. I have tried also hitting up/down and toggle in the UI and this still works even when the dry contact toggle is broken.

[image]

evil-chimera, 2024-08-16:
I have the same issue with my 2 ratgdo's, toggle command shows in logs but the door will not move. 

[image]
[image]

PaulWieland, 2024-09-26:
You can see in both logs that it is triggering the toggle command:
[image]

Followed by the rolling code counter being incremented which indicates that its transmitting the command to the GDO. 

Give that, I'm not sure what the issue could be. Does the door respond to the discrete open or close commands? Does it respond to the toggle button in the web ui?

evil-chimera, 2024-09-27:
So I was having this same problem a while back, this appears to be an issue with the logic that was pushed back in #307. For whatever reason the ratgdo appears to be defaulting to sending the close command after (in my case) the first open close cycle. I looked back at attempts to implement dry contact toggle in the past and found #284, and forked and re-implemented his version of the dry contact logic. I have been running on this slightly modified version of the dry contact toggle logic since I forked it and I have not had any issues. I am new to github and this specifically is the first project I have had any reason to contribute to so I am not familiar with the correct way of submitting this for a proper fix. I have also adjusted some of the delays for my own personal use case. If you would like me to submit this as a pr please let me know. Thanks!

https://github.com/evil-chimera/esphome-ratgdo/commit/fa9852bd1a1d94a6ee5626bfad8ee350338e1ae4

PaulWieland, 2024-09-27:
What makes you think it's sending a close command and not a toggle command?

PaulWieland, 2024-09-27:
Ok, I will try and replicate your test with some some additional logging to see what value is being transmitted.

PaulWieland, 2024-09-27:
The fundamental issue here is that close and toggle are being triggered simultaneously on the leading edge. That's causing two commands to be sent (Close and Toggle), one of which is ignored. 

To fix this, a delay has to be added before evaluating the GPIO's states.

PaulWieland, 2024-09-27:
[Fix is here](https://github.com/ratgdo/esphome-ratgdo/tree/dry_contact_toggle_fix). Requires a little more testing before being merged.

PaulWieland, 2024-09-28:
No, the OTA requires you to upload a compiled firmware file. Unfortunately the github build process which compiles the firmware is [currently broken](https://github.com/ratgdo/esphome-ratgdo/actions/runs/11077344098) and needs to be fixed. 

The only way to run this branch right now is rather convoluted as you would have to clone the repo and build locally. To make matters worse, the yaml files refer to the main branch of the repo, so you have to also know how to deal with that. 

As soon as I'm able to figure out how to fix the github build process, I'll merge the change.

PaulWieland, 2024-09-28:
The automated build process seems to be fixed and the [PR](https://github.com/ratgdo/esphome-ratgdo/tree/dry_contact_toggle_fix) was merged into the main branch.

closed 2024-09-28 by PaulWieland

turbo2ltr, 2024-10-20:
It would be useful if this was added to the wiring diagram. Not knowing this was a feature, I set up an automation in HA but was unhappy with the delay. I couldn't believe that there wasn't method to add a simple button closure without having HA in the middle.  But apparently there is! Going to rewire now.
I actually thought the "Wall mounted button" in the wiring diagram was for the simple dry contact button until I saw a picture and realized it was for the "smart" button which I don't have...which is 1/2 the reason I bought the board.  (HA integration was the other half)

turbo2ltr, 2024-10-20:
Ok the hardware solution is no better than the HA solution, if not a little worse.  Why do I have to hold the button down so long? Surely you can debounce in less than a second.