ratgdo/esphome-ratgdo #319: Using dry contact pins to monitor open/close status

bradyemerson, 2024-08-12:
Hi, I have a basic open/close magnetic sensor attached to the garage door and would like to use the dry contact pins to read the status of the sensor.  Is this possible?

PaulWieland, 2024-08-15:
[Please check the dry contact wiring diagram](https://ratcloud.llc/pages/installation).

closed 2024-08-15 by PaulWieland

bradyemerson, 2024-08-15:
Hi Paul, thanks for the link. On the Security diagram it says that the bottom dry contact pin pulls to ground when the door is closed. If I'm understanding that correctly, I believe that means I can monitor open/close from that pin. I'm looking to do the opposite - - read open / close from an external magnetic sensor. 

The use case is if someone opens the door from the emergency handle, I don't believe the opener knows the door is open. But the magnetic sensor would. 

Is that interpretation correct?