ratgdo/esphome-ratgdo #321: Going offline randomly

angrycatmeowmeow, 2024-08-17:
Ratgdo 2.5i is going offline randomly for random amounts of time.

esphome logs attached.
[logs_ratgdov25i-fada07_logs.txt](https://github.com/user-attachments/files/16644872/logs_ratgdov25i-fada07_logs.txt)

Not mesh client appears in my router logs quite frequently but doesn't always coincide with the device going offline (Asus ET12). Ratgdo still appears in my client list when this happens and shows connected but is unreachable in HA or in a browser. I tried giving it a static IP, changing DNS to quad9, and also binding it to both of my mesh nodes, but none of that helped. I also tried changing the channel width on 2.4 and 5ghz and disabling automatic channel selection on my router. The device is about 8ft away from anything in the garage. This happens on MQTT and ESPhome firmware.

[image]

```
Aug 17 08:44:38 kernel: 08:3A:8D:FA:DA:07 not mesh client, can't delete it
Aug 17 08:46:14 kernel: 08:3A:8D:FA:DA:07 not mesh client, can't delete it
Aug 17 08:48:17 kernel: 08:3A:8D:FA:DA:07 not mesh client, can't update it's ip
Aug 17 08:49:33 kernel: 08:3A:8D:FA:DA:07 not mesh client, can't delete it
```

clumsyCoder00, 2024-08-17:
Mine has just recently disconnecting from homeassistant as well. I just updated to the newest esphome with no change in behavior.

Automan69, 2024-08-27:
> Mine has just recently disconnecting from homeassistant as well. I just updated to the newest esphome with no change in behavior.

Same problem

arjenbos, 2024-09-05:
Since a couple of days I experience the same issue.

PaulWieland, 2024-09-27:
> This happens on MQTT and ESPhome firmware

These are completely different networking stacks. It's not a firmware issue.

closed 2024-09-27 by PaulWieland