ratgdo/esphome-ratgdo #322: Spurious Door Open Messages

johannesparty, 2024-08-24:
ratgdov2.5i, secplus1, LiftMaster 3800.

### Basic

Randomly start getting door opened events; comes in a cluster usually at night.  After a bit, get door closed again.  Might not get any for 24 hours, then get 10-20 times in a few hours.

### Details - other info which might be relevant

- Have myQ hub connected to each garage door also (maybe don't need it anymore; dropped their HomeKit bridge to switch to ratgdo, but left the myQ hub for now as backup/easy warning if left open.
- I have 3 garage doors (1,2,3) with two hubs (hub1: garage 1,2), (hub2: garage 3).
- Garage 2 has backup battery
- All three Garage have 398LM Smart Control Panel and I've turned off motion sensor on all 3
- I do not have any lift master related lights installed
- I do not believe I am losing WiFi connection to any ratgdo (looking at wifi access point data).
- I get "Discard incomplete packet: [38 ...]" on every garage every 10 seconds or so.
- Using HomeBridge to link to HomeKit (separate Linux server).

### Interesting

This problem only occurs with garage door 2.

### Speculation

- loose connection?  doesn't seem that likely.
- related to having two devices connected to single myQ hub?  maybe?
- bad ratgdo module?  maybe? 

### I have attached a few screenshots of the web interface.

Start of open:

[image]

During open:

[image]

A bit later:

[image]

I am happy to do some debugging if you let me know what to do.  Not really familiar with esphome environment but I guess I can learn!

PaulWieland, 2024-08-24:
swap the wall panel between two of the doors. I bet the problem follows the panel. 

This is a known issue that gets reposted quite often. The speculated cause is failing capacitors on the wall panel.

johannesparty, 2024-08-28:
Looks like you are correct. 

I swapped garage 2 and garage 3 wall panels and now the spurious opens happen on garage 3.  I couldn't find any info about bad capacitors on the LM398 and it isn't obvious which capacitors (none seem similar to the often-reported problem capacitors in LM888.  If anybody knows, that would be great.

I did find an LM398 on eBay and will try that.  Otherwise I guess I'll have to switch to LM889LMMC (times three because ... OCD?).

In any case, thanks so much for your help.

closed 2024-08-28 by PaulWieland

drxenonbloom, 2025-06-15:
I would like to add that I had this same issue on my 398LM. I replaced the two largest capacitors at the bottom of the wall panel (they were 25v 220uf) and it seems to have rectified the issue. I'm still getting the "Discard incomplete packet: [38 ...]" warning when the door is closed but no more random open/close events. Just in case anyone else has this issue.