ratgdo/esphome-ratgdo #323: Not automatically being added as a MQTT device in HA.

audioeptesicus, 2024-09-18:
My Home Assistant prefix is the default, homeassistant. I can flash the firmware in ESPhome in HA, and access the IP via my browser and control everything properly. However, I cannot get the ratgdo to appear in HomeAssistant to be able to control it and set it up in my HA dashboard. There is no documentation on this portion of the product.

How can I get my ratgdo v2.5 to add as a device in Home Assistant?  Are there yaml entries necessary for my mqtt username and password? Something else?

restauffer, 2024-09-18:
I think you are confusing the ESP firmware vs the MQTT firmware. If you used ESPHome in HA to flash it then you have the ESP version not the MQTT version. I assume you have the ESPHome menu enabled in your sidebar. Can you see the device there? If you do and it does not show in Devices, see if you have a notice at the top when you open Settings. It would say something about adding that device. If the notice is not there, go to the ESP integration and click Add Device. Put in the IP or hostname and that should add it.

jeeftor, 2024-09-26:
ESPHome != MQTT :)

PaulWieland, 2024-09-26:
And if you are using home assistant, then I would strongly suggest you use the ESP Home firmware and not the MQTT firmware.

closed 2024-09-26 by PaulWieland

audioeptesicus, 2024-09-26:
I used the ESPHome firmware, but the device wasn't being added, and I'm not sure why.  I added the MQTT broker info to the ESPHome yaml file and it finally showed up and I was able to successfully control the garage door.