ratgdo/esphome-ratgdo #326: Static IP causes Ratgdo to disconnect in HA

tshevlin, 2024-10-10:
Kind of a weird issue...I have a TP-Link Omada system and the newest Ratgdo. Whenever I assign the Ratgdo a static IP through Omada, it will go live for a few minutes and then become 'unavailable' in HA. Then throughout the day at random intervals, it will connect for a few minutes and then again become unavailable.

If I remove the static IP, it is completely stable.

Seems like this may be an Omada issue, but not sure. Connectivity doesn't seem to be an issue. There's an AP in the garage, a few feet away from the Ratgdo. The signal and SNR are very strong, and the uptime never drops. I've checked the static ip as well; it's not assigned to any other devices. No power save enabled.

I guess i can keep it as is without a static IP, but every guide online strongly recommends it. 

Any thoughts on why this is happening?

Here's the error in HA logs:
`Can't connect to ESPHome API for ratgdov25i-0af96a @ 192.168.107.18: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.107.18', port=6053))]: [Errno 113] Connect call failed ('192.168.107.18', 6053) (SocketAPIError)`

PaulWieland, 2024-10-10:
FWIW, I have DHCP reservations set on Unifi without any issue.

tshevlin, 2024-10-10:
Gotcha. Maybe I just need to keep it as is without a DHCP reservation. I don't understand why this is happening tho; very annoying lol. 

All my IoT devices have a DHCP reservation and none have this issue.

PaulWieland, 2024-10-10:
After you set the reservation, have you restarted the ratgdo control board? Can you see it in any of the logs on the router?

PaulWieland, 2024-10-10:
When you added the reservation, did you change the IP address? Or did you keep the same one that DHCP gave it?

tshevlin, 2024-10-10:
When I add the reservation, I completely unplug both the ratgdo and the entire network. For your second comment, I haven't messed around at all with the IP settings on the Ratgdo itself. I just do everything through Omada. Once I reboot everything, the Ratgdo pops right up on Omada with the new IP that I reserved for it

PaulWieland, 2024-10-10:
Try setting the reservation to the same IP address that the DHCP server gave it

tshevlin, 2024-10-10:
So it currently is assigned the IP of 192.168.107.56. 

My DHCP range is 192.168.107.50 through 192.168.107.254. If I assign 56, wouldn't that run the risk of another client being assigned the same IP? Or should Omada know not to do that?

Sorry if I'm misunderstanding; I'm not super proficient with networking

PaulWieland, 2024-10-10:
Thats the point of a DHCP reservation. You are reserving it for a device with a specific MAC address. If you are setting the DHCP reservation outside of your configured DHCP range, it might not work (its a _DHCP_ reservation - it makes sense that it would want to use an allowed DHCP address, but this could of course be vendor dependent)

PaulWieland, 2024-10-10:
https://www.tp-link.com/us/support/faq/3200/

Assuming the IP address is pre-filled with the DHCP address, just check the _enable_ checkbox and leave the IP address alone. Their notes show that the behavior is version dependent.

In the event you want a true static ip address and not a dhcp reservation then you have to follow the [ESP Home way of setting a static IP](https://esphome.io/components/wifi.html).

closed 2024-10-10 by PaulWieland