ratgdo/esphome-ratgdo #330: RJ070, 2.53 board, garage door won't close.

rborkow, 2024-10-27:
Hi - After a few months of normal use I started having an issue where with the GDO board installed the garage door would refuse to close no matter if I tried to use the GDO or the wall panel to close the door. Removing the GDO resolved the issue, and I purchased a new one thinking something may have happened to the board. However, I have the new board installed and am having the same issue. Behavior is the door will start to close momentarily, then stop itself almost immediately.

 Appreciate any insights into what might be going on here!

`
11:11:49 | [D] | [cover:076] | 'Door' - Setting
-- | -- | -- | --
11:11:49 | [D] | [cover:084] | Position: 0%
11:11:49 | [D] | [ratgdo_secplus2:463] | Collision detected, waiting to send packet
11:11:49 | [D] | [ratgdo_secplus2:466] | Collision detected, waiting to send packet
11:11:49 | [D] | [number:012] | 'Rolling code counter': Sending state 48.000000
11:11:51 | [D] | [ratgdo:088] | Door state=OPEN
11:11:51 | [D] | [ratgdo:197] | Light state=ON
11:11:51 | [D] | [ratgdo:203] | Lock state=UNLOCKED
11:11:51 | [D] | [ratgdo:182] | Learn state=INACTIVE
11:11:52 | [D] | [ratgdo:088] | Door state=OPEN
11:11:52 | [D] | [ratgdo:197] | Light state=ON
11:11:52 | [D] | [ratgdo:203] | Lock state=UNLOCKED
11:11:52 | [D] | [ratgdo:182] | Learn state=INACTIVE
11:11:55 | [D] | [ratgdo:088] | Door state=OPEN
11:11:55 | [D] | [ratgdo:197] | Light state=ON
11:11:55 | [D] | [ratgdo:203] | Lock state=UNLOCKED
11:11:55 | [D] | [ratgdo:182] | Learn state=INACTIVE
11:11:58 | [D] | [cover:076] | 'Door' - Setting
11:11:58 | [D] | [cover:084] | Position: 0%
11:11:58 | [D] | [number:012] | 'Rolling code counter': Sending state 49.000000
11:12:00 | [D] | [ratgdo:088] | Door state=OPENING
11:12:00 | [D] | [ratgdo:197] | Light state=ON
11:12:00 | [D] | [ratgdo:203] | Lock state=UNLOCKED
11:12:00 | [D] | [ratgdo:182] | Learn state=INACTIVE
11:12:00 | [D] | [cover:170] | 'Door' - Publishing:
11:12:00 | [D] | [cover:173] | Position: 100%
11:12:00 | [D] | [cover:186] | Current Operation: OPENING
11:12:00 | [D] | [switch:012] | 'Status door' Turning ON.
11:12:01 | [D] | [ratgdo:088] | Door state=OPEN
11:12:01 | [D] | [ratgdo:537] | Cancelling position callbacks
11:12:01 | [D] | [ratgdo:197] | Light state=ON
11:12:01 | [D] | [ratgdo:203] | Lock state=UNLOCKED
11:12:01 | [D] | [ratgdo:182] | Learn state=INACTIVE
11:12:01 | [D] | [cover:170] | 'Door' - Publishing:
11:12:01 | [D] | [cover:173] | Position: 100%
11:12:01 | [D] | [cover:186] | Current Operation: IDLE
11:12:01 | [D] | [switch:012] | 'Status door' Turning ON.
11:12:03 | [D] | [ratgdo:088] | Door state=OPEN
11:12:04 | [D] | [ratgdo:197] | Light state=ON
11:12:04 | [D] | [ratgdo:203] | Lock state=UNLOCKED
11:12:04 | [D] | [ratgdo:182] | Learn state=INACTIVE
11:12:06 | [D] | [esp8266.preferences:238] | Saving preferences to flash...

`

rborkow, 2024-10-27:
Tightened up my obstruction sensor wiring and it's resolved.

closed 2024-10-27 by rborkow

PaulWieland, 2024-10-27:
99% of the time when a door starts to close and then reverses its either because the obstruction sensors are not connected or are detecting an obstruction. Most chamberlain GDOs will flash the on board light on and off repeatedly indicating that there is an obstruction issue.

The other 1% of the time its because the door is physically obstructed, either by jamming in the track or by an object.