ratgdo/esphome-ratgdo #334: Light button on wall panel no longer works after installing ratgdo

chmeeedalf, 2024-11-04:
I have two Security+ 1.0 LiftMaster garage door openers with the 3-button wall panel (open, lock, light).  After installing my new ratgdo devices on the Light button no longer works.  I don't know about the lock button, haven't tested that.  However, the light control from the ratgdo does work, and the main Open button works, which makes this rather confusing.

Some details:
Firmware: 2024.9.2 (Oct 14 2024, 04:22:30)
Hardware: 2.5i 

and:

Firmware: 2024.10.2 (Oct 28 2024, 04:23:29)
Hardware: 2.5i

PaulWieland, 2024-11-04:
Please check the FAQ. https://ratcloud.llc/pages/frequently-asked-questions

closed 2024-11-04 by PaulWieland

chmeeedalf, 2024-11-04:
Oops, sorry for the noise.  I thought I had read through the FAQ, and clearly missed that one.