ratgdo/esphome-ratgdo #336: 889LM goes into boot loop whenever a command is issued

depoll, 2024-11-07:
I have 3 GDOs that I'm setting up ratgdo for.  For my 2 Chamberlain 2000SDR GDOs with 889LMs on the wall, my ratgdo (2.53i hardware, Security+ 1.0) sees the status from the panel and can issue a single command, after which the 889LM on the wall goes into a boot loop (blinking light, 3 short beeps after some time, then it repeats itself).  In the logs, I can see that the moment the command is issued, I stop getting status updates from the 889LM (presumably because it's rebooting), and can only restore things if I unplug the ratgdo until the panel has booted before plugging back in.

Any idea why this might be happening or how it might be fixed?

depoll, 2024-11-07:
One more note -- here's the last log line I get when the 889LM goes offline:

```
15:16:56 | [W] | [ratgdo_secplus1:264] | [453045] Discard incomplete packet: [38 ...]
```

PaulWieland, 2024-11-08:
I think the capacitors on your wall panel are failing and it’s browning out. That’s the only reason i can think of that would make it reboot.

depoll, 2024-11-08:
Strange that it would happen on both of them, though. I suppose I can try replacing one and see if it fixes the problem.

depoll, 2024-11-12:
I just tried again with brand new 889lm units and both still have the same problem. It's almost as though it's cutting power to the wall panel briefly or something like that.

depoll, 2024-11-12:
https://github.com/ratgdo/mqtt-ratgdo/issues/68 looks like the same behavior, fwiw.

depoll, 2024-12-13:
Any other ideas here?  I'm fairly stuck on these two units at the moment.  I can use them to see status, but making any change (e.g. toggling the lights, opening/closing the door) causes this to happen pretty consistently.

jdranchman, 2024-12-13:
> Any other ideas here? I'm fairly stuck on these two units at the moment. I can use them to see status, but making any change (e.g. toggling the lights, opening/closing the door) causes this to happen pretty consistently.

I'm pretty sure that controller is the same as the Chamberlain one. The electrolytic caps (2) are too small and they can easily be degraded over time IMHO. I replaced them with double the capacitance and I haven't had that issue in over a month - even on my older built controller from 2013.  I did buy a new one just in case made 03/2024.

If that didn't work for me I was ready to ditch the smart control panel and just get the button or button/motion sensor. You don't really need the screen to do anything after you get it up on HA.

depoll, 2024-12-13:
@jdranchman Are you talking about the capacitor in the wall panel or something in the GDP itself?

jdranchman, 2024-12-13:
> @jdranchman Are you talking about the capacitor in the wall panel or something in the GDO itself?

If you tear apart the wall control panel, there are two electrolytic caps. Both of those need swapped out. They may be enough for the device itself but not have enough energy when the ratgdo uses the comm line.

depoll, 2024-12-13:
Interesting.  I've already swapped for brand new control panels, so I know mine aren't old.  I suppose I could try swapping the capacitors for larger ones anyway.

PaulWieland, 2024-12-13:
Closing this as it is a known issue with a known resolution.

closed 2024-12-13 by PaulWieland