ratgdo/esphome-ratgdo #339: Issues with chamberlain opener, ratgdo and wall switch not responding on first press

inittabayanami, 2024-11-13:
I have 3 Chamberlain B2211TMC openers, each with a ratgdo v2.5 kit, nothing additional hooked up beyond the door sensors and the included push button
ratgdo's are running latest firmware Firmware: 2.5i (ESPHome 2024.10.3) on ESPhome, control is usually done via homeassistant to open/close doors.
The issue i'm running into is if I open the doors through home assistant then go to use the physical push button on the wall I will usually have to push it twice to get it to respond. If I do not use homeassistant to open the door and only the car remotes/wall remote I do not have this issue

The problem is this doesn't happen every time, I just open and closed the door multiple times in home assistant and then hit the wall button and it worked perfectly. but when I went out there to test I hit the wall button initially and it did not open, I hit it again and then it opened. I cannot be sure but there seems to be a time component involved

Any sugggestions on how to track this down?

greggitter, 2024-11-14:
You're probably referring to this....?  https://github.com/ratgdo/esphome-ratgdo/issues/253

inittabayanami, 2024-11-14:
> You're probably referring to this....? #253

I did see this when I was looking but didn't think it was related (could be wrong here). all 3 garage door openers experience this issue, and all 3 are new installed at the same time as the ratgdo's. 
The wall controls are also not 'fancy' they do communicate via security 2.0 (yellow learn button) but beyond that are just a button to open/close, and a light control (model: 041A7367-3)

inittabayanami, 2024-11-14:
Okay will post over there too

netguy101, 2024-12-02:
> Okay will post over there too

inittabayanami, I am about to buy the ratgdo and googled it along with my garage door model (B2211T) to check if supported it. While it’s not the exact model as yours, I have had it installed since January of this year (with no ratgdo involved) and I’ve had this random double click issue regularly. I’ve just gotten used to it but just an FYI that I doubt ratgdo is the issue.

inittabayanami, 2026-10-27:
Disclaimer, This 'fix' was generated via claude code after I got tired of having to double press the wall button all the time. it took a look at the logs while I hit the button, noticed the opener was ignoring the first press and added the following to the ratgdo's configuration yaml's, so far it seems to work pretty well, when the opener button bugs out there is a very short delay, then the opener actually starts to move rather than me having to press the button again. maybe something like this could be added to the ratgdo code itself if it sees the wall control trigger a toggle but the opener not respond?

```
# Wall-button retry (added 2026-10-27, esphome-ratgdo #339): the Sec+2 wall button
# is sometimes ignored by the opener after another device moved the door. If the
# ratgdo sees a wall press and the door hasn't started moving within 1.2s, send
# our own toggle (which the opener always accepts).
binary_sensor:
  - id: !extend ratgdov25i_button
    on_press:
      - script.execute: wall_press_retry

script:
  - id: wall_press_retry
    mode: single
    then:
      - if:
          condition:
            lambda: |-
              return millis() > 30000 &&
                     id(ratgdov25i_garage_door).current_operation == COVER_OPERATION_IDLE;
          then:
            - delay: 1200ms
            - if:
                condition:
                  lambda: 'return id(ratgdov25i_garage_door).current_operation == COVER_OPERATION_IDLE;'
                then:
                  - logger.log:
                      level: WARN
                      format: "Wall press ignored by opener, sending toggle"
                  - lambda: 'id(ratgdov25i).door_toggle();'
````