ratgdo/esphome-ratgdo #345: Inability to import ratdgo disco into ESPHome Compiler

darth-monkey, 2024-12-08:
Hello all,

I was able to succesfully flash my newly received ratgdo disco and add to HA as an ESP device; however am unable to import it into the ESPHome compiler with the following error:

Is this expected at this time - and is there any guidance wether or not this HA add-on should be used for OTA management/log viewing and such? 

Thank you! 

```
INFO ESPHome 2024.11.3
INFO Reading configuration /config/esphome/ratgdo32disco-ce00d8.yaml...
Failed config

packages: [source /config/esphome/ratgdo32disco-ce00d8.yaml:5]
  
  Failed to load packages. v32disco_secplusv1.yaml does not exist in repository.
  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v32disco_secplusv1.yaml@main
```

khicks, 2024-12-14:
I was able to get the compiler working by changing the branch to `ratgdo32`. Whether this is the recommended approach, I'm unsure, but it'll do for now.

```yaml
packages:
  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v32disco.yaml@ratgdo32
```

darth-monkey, 2024-12-16:
> I was able to get the compiler working by changing the branch to `ratgdo32`. Whether this is the recommended approach, I'm unsure, but it'll do for now.
> 
> ```yaml
> packages:
>   ratgdo.esphome: github://ratgdo/esphome-ratgdo/v32disco.yaml@ratgdo32
> ```

Thank you @khicks ! This was very helpful and seems to have done the trick. Might just be checking the main branch periodically to see if the board comes up there. 

**Warning for those as inatentive as me** If you are using Security 1.0 doors.. make sure to reference the right file 🙃 

```packages:
  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v32disco_secplusv1.yaml@ratgdo32
```

Found myself with the door cracked opened in the morning after flashing it in the evening :)

khicks, 2024-12-16:
> Found myself with the door cracked opened in the morning after flashing it in the evening :)

@darth-monkey Ooof, sorry about that. 😬 I copied my own config without checking what you wrote. Good catch.

darth-monkey, 2024-12-16:
> > Found myself with the door cracked opened in the morning after flashing it in the evening :)
> 
> @darth-monkey Ooof, sorry about that. 😬 I copied my own config without checking what you wrote. Good catch.

@khicks  - that one's totally on me :) i've grown complacent with HA operations and really should've physically confirmed i can still operate the device and nothing changed. Ah well - lesson learned :)

e28eta, 2025-02-11:
I think this is fixed now that #337 and #338 have been merged