ratgdo/esphome-ratgdo #353: Open/Closed entity in Homeassistant

ba05, 2024-12-29:
Is it possible to add an open/closed entity in home assistant? If I click on the door button it says "Closed" or "Open" but I can't access this through an entity.

doctorkb, 2024-12-29:
The `cover` entity should have that as the state...

ba05, 2024-12-31:
```cover``` has it as a state, but this cannot directly be added to an entity card without creating a binary sensor using the close/open value.

doctorkb, 2024-12-31:
You could use a template card. Or, if you're that set on a binary sensor, just set up a helper.

tmason818, 2025-01-11:
I use this for a  simple Open/Close toggle:
https://www.home-assistant.io/integrations/cover.template/

This is the code I added to my configuration.yaml.

```
#Virtual Garage Door control entity for "toggle" up/down Toggle control
cover:
  - platform: template
    covers:
      garage_door:
        device_class: garage
        friendly_name: "Virtual Garage Door Control"
        value_template: "{{ states('cover.garage_door_ratgdo_21719_door') }}"
        open_cover:
          action: cover.open_cover
          entity_id: cover.garage_door_ratgdo_21719_door
        close_cover:
          action: cover.close_cover
          entity_id: cover.garage_door_ratgdo_21719_door
``` 

You'll end up with a new entity you can add to cards:

[image]