ratgdo/esphome-ratgdo #357: Vehicle distance actual is NA

iissmart, 2025-01-11:
ratgdo disco running firmware 2024.12.2 Dec 31 2024, 20:29:40. It doesn't seem to be reporting a Vehicle distance for some reason. On the ratgdo device webpage the log just shows periodic voltage and wifi signal. I have another ratgdo disco that reports the vehicle distance, so I'm not sure what might be different with this one. Any idea how to troubleshoot this?

iissmart, 2025-01-11:
Maybe it's just not getting a good reflection back from the vehicle. Does it need to be a perpendicular surface that it's aimed at? The one that works is a single stall where the gdo is directly above the car. The one that doesn't work is a double stall where the gdo is right at the edge of the side of the vehicle. What are some good ways to mount it for a double stall gdo?

PaulWieland, 2025-01-11:
A value of NA means it hasn't reported any valid measurement yet, which normally will not happen. Make sure there isn't a yellow sticker stuck over the sensor. Next hold a piece of paper about 30cm (1') away from the sensor to check that the sensor is working. Alternatively you can aim the sensor at a ceiling or wall (sheetrock surfaces work well). 

> Does it need to be a perpendicular surface that it's aimed at?

No, it does not need a perpendicular surface. Most overhead door installs have the opener over the roof of the car. On vehicles with a glass roof, the laser has to be aimed forward at the hood of the car (this is how mine are installed).

iissmart, 2025-01-11:
There is something that looks like kapton tape over what I'm assuming is the sensor on both ratgdos, am I supposed to remove that? I left it on both, and the ratgdo in the single stall is still working perfectly.

Actually, now that the vehicle is gone from the double stall, the ratgdo started reporting a distance as I'd expect. I do think it's just a positioning/aiming issue for this one, but I'll have to wait a few hours for the vehicle to come back before I can try again (still curious about the kapton tape though).

iissmart, 2025-01-11:
OK, removed the tape from both and re-positioned the ratgdo, and it's now reporting the position correctly. Thanks for the help!

You may want to add some documentation about removing the kapton tape from the sensors, maybe on https://ratcloud.llc/pages/mounting . I've had a few devices that *looked* like there was a temporary film on the screen, only to find out that it was meant to be permanent and damaged the device when I started to peel it off, so I was playing it safe with this one.

closed 2025-01-11 by PaulWieland