ratgdo/esphome-ratgdo #359: After last update it no longer shows opening/closing states.

jimz011, 2025-01-20:
After the last update I am no longer receiving position updates for my door. It will show open after the opening duration has passed and the same for close. But nothing in between.

I did not change the configuration and I have tried downgrading and even upgrading to dev firmware to see if it would resolve it. Also whenever the device restarts it will show the state as "open" despite being closed.

I also removed the device from HA and reinstalled it. And after that I opened and closed it fully multiple times. But nothing changed. I am guessing it is a problem with the config since the logs don't output position changes in esphome itself.

I am a bit lost what do try next.

PaulWieland, 2025-01-20:
Recalibrate your opening and closing durations.

closed 2025-01-20 by PaulWieland

jimz011, 2025-01-20:
I am really sorry to bother you but how does one achieve that? I have trouble finding this in the docs. It was exactly the calibration function I couldn't find hence the opened issue.

Also it states that I need to fully open/close to calibrate it. But I did this many times without success.

PaulWieland, 2025-01-20:
Try setting both to zero and then cycle the door.