ratgdo/esphome-ratgdo #361: When CLOSE command is sent Door remains OPEN but reports CLOSED state briefly | Security+ 1.0 | ESPHome | 2.5i Board on 2024.10.3 Firmware

tmason818, 2025-01-23:
Using wall mount opener RJ020 **Security+ 1.0** device.

I've been trying to catch this bug for a few months now because it is not common. Every so often sending a CLOSE command will not close the Garage Door but RATGDO briefly reports it as being in the CLOSED status.

I'm never able to catch this in the RATGDO logs as they scroll away too quickly so the best I can do is show how it is logged in Home Assistant:

[image]

At 5:27:41 AM the CLOSE command was sent via Home Assistant but isn't logged and the door doesn't physically close.
Even so 15 seconds later at 5:27:16 AM the RATGDO reports the Door is in the CLOSED state but there is no log for the transitive CLOSING state.
Then, only 1 Second later, at 5:27:17 AM the RATGDO reports the Door is now in the OPEN state- also without any logs of a transitive OPENING state.

For reference my Garage Door takes 13 seconds to Open or Close as can be seen after the another CLOSE command at 5:33:10 AM. The obstruction sensors were clear.

I have had similar issues with the OPEN command but haven't been able to catch it in the logs.

xmlguy74, 2025-02-08:
I'm seeing the exact same behavior.

panzerwaffen, 2025-03-26:
Just adding to the list same problem.  About ready to go back to my Shelly 1's at they were reliable.

tmason818, 2025-03-26:
> Just adding to the list same problem.

What's strange is I don't recall this being an issue when using the MQTT firmware . But I believe that firmware is being deprecated with this one being the replacement?

PaulWieland, 2025-03-26:
Are you using wall panel emulation mode? If not, what wall panel do you have? How often does the error occur?

panzerwaffen, 2025-03-26:
> Are you using wall panel emulation mode? If not, what wall panel do you have? How often does the error occur?

889LM purchased directly from Chamberlain (I add this I purchased two 889LM's  from Amazon that were fakes after my 888LM's died), issue was not observed on a competitor device's (but they had enough other issues they are in my IOT graveyard). 

Generally my  leaving close door automation works. 

Arrival close door automation quit working last year, but it is like the command from HA is ignored because I don't get the double notification. I have tried switching from call service to just cover down.  No change.

Now my open door automation has been flaky lately. 

All automation's in HA show they are working. 

Everything else works correctly.

Updated firmware via USB in mid February.   

Thank you

tmason818, 2025-03-26:
> Are you using wall panel emulation mode? If not, what wall panel do you have? How often does the error occur?

I am using the wall panel that came with my RJO20 which is listed in the manual as:

**Door Control (myQ® Control Panel) Model 041A7928-3**

It has _Chamberlin_  and _myQ_ branding but looks very similar to the 889LM?

tmason818, 2025-03-28:
Is there a better way to get logs directly from the RATGDO device? The logs scroll so fast its nearly impossible to catch this issue in the logs.  This pic is just 3 seconds of logging. Its very noisy. 

[image]

Rev-777, 2025-05-18:
Identical issue here.

Rev-777, 2025-07-13:
Any update to this issue? Still have the same problem.

Rev-777, 2025-11-29:
I went back to MQTT because there was no update/fix on this.

Good luck with the project.

tmason818, 2025-11-29:
> I went back to MQTT because there was no update/fix on this.
> 
> Good luck with the project.

That's such a funny coincidence....I did the exact same thing yesterday.

I attribute the problem to the RJ020 just being an oddball device. After actually using ESPHOME with the RATGDO I didn't really need any of the extra features it provided. Using MQTT suites my needs wonderfully and I remember the reliability being perfect (aside from the strange issue that OPENS the garage door even when the close door command is sent to it after a power outage-  but that issue was present in ESPHOME as well and is easy enough to work around).

Also I could also never figure out how to password protect the ESPHOME control panel while doing so with MQTT was easy as pie.

I'll keep this issue open for now to see if the issue does happen to occur with MQTT but it's up to @panzerwaffen and @xmlguy74 to provide any further troubleshooting.