ratgdo/esphome-ratgdo #369: Obstruction Detection Method

Daz42, 2025-02-10:
Hello, does anyone know what to change in the latest ratgdo ESPHome to set the obstruction method to status instead of the sensors (pins on the board). The modification used to be in ratgdo.cpp but this appears to have changed in later versions.

PaulWieland, 2025-02-11:
Unhooking the black obst wire should be all you need to do.

Daz42, 2025-02-11:
Hi Paul, thanks - I guess you built in that feature a while ago and that means that I will no longer have to modify the firmware. 

I've disconnected the black wire and now the obstruction sensors are connected directly to the GDO. However, there is still a problem. All functions as it should - door opens/closes and is more responsive than my previous firmware! The problem is when I obstruct the sensors, the ratgdo DOES detect the obstruction but immediately afterwards drops off the network for 5-10 seconds (pings are dropped) before appearing back on the network again. This is consistent and the problem can be reproduced every time.

By the way, this is with only one pair of sensors connected.

I can take a video of the problem if that helps?

Daran.

Daz42, 2025-02-13:
Ok, did some more testing and even thought I had a bad ratgdo so I tried my spare ratgdo - both the same result with the latest firmware. I moved the router closer to the ratgdo (a couple of feet away), turned off any IOT devices nearby but this didn't help. Also changed the USB cable.

The ratgdo works perfectly, opens and closes the door etc and you can see the web log ticking over smoothly. The problem arises with you obstruct the obstruction sensors - which is detected - (just a single pair connected) and the ratgdo drops 2-4 pings before coming back which happens consistently. The black obstruction wire is NOT connected so it should be reading the status packet. Given that the old firmware that I modified last year in May still works, there has to be an issue with the new firmware Paul? What would be causing the ESP to drop pings?

Daz42, 2025-02-14:
Found the problem and hopefully this helps someone else. This was quite a tricky one. The issue was a faulty router - an Asus RT-AX68S which was a mesh node to another Asus RT-AX68S. I swapped the faulty mesh node with a vanilla ZTE router and the ratgdo stayed on the network. The obstruction sensors worked - even using 2 pairs of them connected with the latest firmware. The only small issue I am still having is that I have to query the ratgdo every second with Home Assistant to see the obstructions sensors update but I've had this issue since May 2024. Not sure if Paul would like to look at this one for a future release of the firmware. For now I've ordered a new Asus router so fingers crossed this solves the issue.

closed 2025-02-14 by Daz42