ratgdo/esphome-ratgdo #370: Change WiFi Network

vmcosco, 2025-02-12:
I just received my ratgdo32 a week ago.  Setup went perfect: I connected to device WIFI, set up my main WIFI, downloaded the ESPHome firmware and installed/updated and the device popped right up in Home Assistant.

I would like to move the device to an IoT WiFi that is on an IoT VLAN (I should have done this from that start, but wanted to make sure it worked).  First, will the device still work if it is on a different VLAN from Home Assistant.  Second, is there a way to change the WiFi connection? 

I have had suggestions to reflash the firmware but since the new OTA method does the WiFi connection first, that doesn't give me the option to change networks.  I had suggestions to use the ESPHome Builder in HA but didn't want to wreck anything or unnecessarily take the device off the opener if there is an OTA way.

PaulWieland, 2025-02-12:
captive portal is enabled by default.
https://esphome.io/components/captive_portal.html

vmcosco, 2025-02-12:
So should I turn off the current Wifi to force the captive portal and then change the Wifi?  I haven't done anything with yaml config so not sure how to manage that.

vmcosco, 2025-02-13:
Thanks for the detailed reply.  I have the IoT WiFi up and running with several non-human devices on it.  It works very well like you mention.

> For ESPHome devices like the RATGDO, that are successfully registered in ESPHome, you can extract, view and edit the YAML for it, edit it and push out the new firmware with a new primary Wifi connection on OTA

This is part of my issue as I am not sure how to do this.  RATGDO is my first ESPHome device.  Do I use the ESPHome Builder in Home Assistant?  I do not know of any other way "extract, view and edit the YAML"

vmcosco, 2025-02-13:
Thank you again for the detail

> For now, the left Nav bar, as an admin, you should see ESPHome or ESPHome Compiler and you will click into and see the devices managed by ESPHOME. You should see RatGDO.

I have seen that mt RATGDO is "Discovered", I assume that I should then click "Take Control".  When I do that, I get a popup window allowing me to change device name and also asking me enter WiFi credentials.  Should I enter the credentials for the new IOT Wifi that I want to connect to?  If I do that, would I even need to do the YAML edit?

benze, 2025-03-12:
@vmcosco Did you ever manage to get everything working?  I'm pretty much in the same spot you were when you posted a month ago.