ratgdo/esphome-ratgdo #373: Vehicle Distance Actual

Idowza, 2025-02-19:
I recently installed the Disco and I'm really enjoying it. Everything is working perfectly except for the "Vehicle Distance Actual" sensor. It's fluctuating between 750 and 780mm, but then occasionally jumps to 4500mm for a second or two. This jump also causes the "Vehicle Detected" status to switch from ON to OFF. Any ideas what might be causing this or how to fix it?

[image]

PaulWieland, 2025-02-19:
what's it pointed at?

Idowza, 2025-02-19:
It's pointing straight down at the roof of my car.

PaulWieland, 2025-02-19:
Can you post a picture?

Idowza, 2025-02-19:
[image]
[image]
[image]

Idowza, 2025-02-19:
> I need to see the roof of the car in the photo

Sorry I had to edit the pic, it's up now.

PaulWieland, 2025-02-19:
No problem. It's hard to tell from the picture, but does the car have a rear sunroof?

Idowza, 2025-02-19:
It does not.

PaulWieland, 2025-02-19:
Ok, everything seems pretty normal actually.  All automations should be built off of the arriving and leaving sensors, which only change state within short window of a door state change. The vehicle detection sensor is more of an aid for initial setup.

The 4500 measurement is the max distance which is reported when the sensor is receiving out of range data. The sensor measures dozens of times per second, and 30 sequential readings are required to be `all in` or `all out` for the Vehicle Detected state to change.  I've had to make a choice to strike a balance between speed and accuracy. If we increase the filtering, the accuracy will go up but the time it takes to detect a state change will also increase. I'm still adjusting this balance. 

So as long as your arriving and leaving states are accurate then there isn't a problem to worry about.

[image]

[image]

Idowza, 2025-02-19:
Thank you for your help and clarification.

PaulWieland, 2025-02-19:
No problem

closed 2025-02-19 by PaulWieland