ratgdo/esphome-ratgdo #374: Obstructions not recognized

buddhra, 2025-02-24:
After many months of working correctly, my obstruction sensor has stopped reporting reliably in Home Assistant and on the ratgdo web server page.

I have periods of time in my logs where it shows obstructed for 8 hours at a time when there was no obstruction and other times where I obstruct the door sensors, but the status remains clear. In some cases it responds, but can take minutes to show a problem or clear.

If I disconnect the black wire at the opener, ratgdo immediately updates to problem and immediately clears when I reconnect.

When I test obstructing and clearing the sensor, the status on the wall display updates as expected. 

This change my have been caused by a recent update of the ratgdo 2.5 and ESP32 D1 Mini to ESPHome 2025.2.0. 

I'd appreciate any guidance you can give to further troubleshoot the issue.

jdranchman, 2025-02-24:
I've noticed that I need to clean off my obstruction sensor's lenses every
few months. Dust covers stuff because the doors are not hermetically sealed.
They can also easily get bumped or vibrated out of alignment - which is
another common problem. Once I can successfully test using the LEDs on the
sensors then I move on to whatever they are directly connected to (a
ratgdo in my case).

buddhra, 2025-02-24:
> I've noticed that I need to clean off my obstruction sensor's lenses every
> few months. Dust covers stuff because the doors are not hermetically sealed.
> They can also easily get bumped or vibrated [...]

Thanks, but that doesn't seem to be the issue in my case. The sensors are working correctly based on the led on the sensor and the display on the wall updating as expected.

buddhra, 2025-03-06:
This seems to have been a hardware failure on my ratgdo2.5. I replaced it with a ratgdo32 and the obstruction sensors are now working correctly.

closed 2025-03-06 by buddhra

buddhra, 2025-09-24:
> [@buddhra](https://github.com/buddhra): Can you describe the hardware failure? I'm investigating an issue on mine - with the MQTT firmware I had no issues with the obstruction sensors... but I decided to swap over to ESPHome and now notice inconsistent reporting from the sensor...

Sorry, what I posted above is all I recall at this point. I just replaced it and moved on.