ratgdo/esphome-ratgdo #376: Can't set a specific cover position via HA - 2.5i (ESPHome 2025.2.0)

csobrinho, 2025-02-25:
HI, I've noticed this a few months ago but initially when I installed the device I was able to set the cover of the garage to 10, 15, xx% and the garage door would just move to this position and stop.

Now I can only open/close via the ratgdo or interrupt the door via the contact. Was wondering what type of logs do you need to help debug this issue. Much appreciated!

PS: I'll add logs when the cover position command is called later today.

```yaml
substitutions:
  name: ratgdo
  friendly_name: ratgdo
packages:
  ratgdo.esphome: github://ratgdo/esphome-ratgdo/v25iboard.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: xxxxxxxxxxx
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

```
INFO ESPHome 2025.2.0
INFO Reading configuration /config/ratgdo-ca08fd.yaml...
INFO Updating https://github.com/ratgdo/esphome-ratgdo@None
INFO Updating https://github.com/ratgdo/esphome-ratgdo@main
INFO Detected timezone 'America/Los_Angeles'
INFO Starting log output from 10.10.1.28 using esphome API
INFO Successfully connected to ratgdo @ 10.10.1.28 in 0.007s
INFO Successful handshake with ratgdo @ 10.10.1.28 in 2.645s
[13:19:52][I][app:100]: ESPHome version 2025.2.0 compiled on Feb 25 2025, 11:53:26
[13:19:52][I][app:102]: Project ratgdo.esphome version 2.5i
[13:19:52][C][wifi:600]: WiFi:
[13:19:52][C][wifi:428]:   Local MAC: 48:3F:DA:CA:08:FD
[13:19:52][C][wifi:433]:   SSID: [redacted]
[13:19:52][C][wifi:436]:   IP Address: 10.10.1.28
[13:19:52][C][wifi:439]:   BSSID: [redacted]
[13:19:52][C][wifi:441]:   Hostname: 'ratgdo'
[13:19:52][C][wifi:443]:   Signal strength: -59 dB ▂▄▆█
[13:19:52][C][wifi:447]:   Channel: 1
[13:19:52][C][wifi:448]:   Subnet: 255.255.255.0
[13:19:52][C][wifi:449]:   Gateway: 10.10.1.1
[13:19:52][C][wifi:450]:   DNS1: 10.10.1.2
[13:19:52][C][wifi:451]:   DNS2: 10.10.1.1
[13:19:52][C][logger:177]: Logger:
[13:19:52][C][logger:178]:   Max Level: DEBUG
[13:19:52][C][logger:179]:   Initial Level: DEBUG
[13:19:52][C][logger:181]:   Log Baud Rate: 115200
[13:19:52][C][logger:182]:   Hardware UART: UART0
[13:19:52][C][ratgdo.number:024]: RATGDO Number 'Rolling code counter'
[13:19:52][C][ratgdo.number:024]:   Unit of Measurement: 'codes'
[13:19:52][C][ratgdo.number:028]:   Type: Rolling Code Counter
[13:19:52][C][ratgdo.number:024]: RATGDO Number 'Opening duration'
[13:19:52][C][ratgdo.number:024]:   Unit of Measurement: 's'
[13:19:52][C][ratgdo.number:030]:   Type: Opening Duration
[13:19:52][C][ratgdo.number:024]: RATGDO Number 'Closing duration'
[13:19:52][C][ratgdo.number:024]:   Unit of Measurement: 's'
[13:19:52][C][ratgdo.number:032]:   Type: Closing Duration
[13:19:52][C][ratgdo.number:024]: RATGDO Number 'Client ID'
[13:19:52][C][ratgdo.number:026]:  Type: Client ID
[13:19:52][C][switch.gpio:068]: GPIO Switch 'Status door'
[13:19:52][C][switch.gpio:090]:   Restore Mode: always OFF
[13:19:52][C][switch.gpio:031]:   Pin: GPIO16
[13:19:52][C][switch.gpio:068]: GPIO Switch 'Status obstruction'
[13:19:52][C][switch.gpio:090]:   Restore Mode: always OFF
[13:19:52][C][switch.gpio:031]:   Pin: GPIO15
[13:19:52][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Dry contact open'
[13:19:52][C][gpio.binary_sensor:016]:   Pin: GPIO14
[13:19:52][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Dry contact close'
[13:19:52][C][gpio.binary_sensor:016]:   Pin: GPIO12
[13:19:52][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Dry contact light'
[13:19:52][C][gpio.binary_sensor:016]:   Pin: GPIO0
[13:19:52][C][light:092]: Light 'Light'
[13:19:52][C][version.text_sensor:021]: Version Text Sensor 'Firmware Version'
[13:19:52][C][version.text_sensor:021]:   Icon: 'mdi:new-box'
[13:19:52][C][ratgdo:104]: Setting up RATGDO...
[13:19:52][C][ratgdo:105]:   Output GDO Pin: GPIO5
[13:19:52][C][ratgdo:106]:   Input GDO Pin: GPIO4
[13:19:52][C][ratgdo:107]:   Input Obstruction Pin: GPIO13
[13:19:52][C][ratgdo_secplus2:059]:   Rolling Code Counter: 14145
[13:19:52][C][ratgdo_secplus2:060]:   Client ID: 1488185
[13:19:52][C][ratgdo_secplus2:061]:   Protocol: SEC+ v2
[13:19:52][C][ratgdo.sensor:059]: RATGDO Sensor 'Openings'
[13:19:52][C][ratgdo.sensor:059]:   State Class: ''
[13:19:52][C][ratgdo.sensor:059]:   Unit of Measurement: 'openings'
[13:19:52][C][ratgdo.sensor:059]:   Accuracy Decimals: 0
[13:19:52][C][ratgdo.sensor:059]:   Icon: 'mdi:open-in-app'
[13:19:52][C][ratgdo.sensor:061]:   Type: Openings
[13:19:52][C][ratgdo.sensor:059]: RATGDO Sensor 'Paired Devices'
[13:19:52][C][ratgdo.sensor:059]:   State Class: ''
[13:19:52][C][ratgdo.sensor:059]:   Unit of Measurement: ''
[13:19:52][C][ratgdo.sensor:059]:   Accuracy Decimals: 0
[13:19:52][C][ratgdo.sensor:059]:   Icon: 'mdi:remote'
[13:19:52][C][ratgdo.sensor:063]:   Type: Paired Devices
[13:19:52][C][ratgdo.lock:012]: RATGDO Lock 'Lock remotes'
[13:19:52][C][ratgdo.lock:013]:   Type: Lock
[13:19:52][C][ratgdo.switch:068]: RATGDO Switch 'Learn'
[13:19:52][C][ratgdo.switch:070]:   Icon: 'mdi:plus-box'
[13:19:52][C][ratgdo.switch:090]:   Restore Mode: always OFF
[13:19:52][C][ratgdo.switch:014]:   Type: Learn
[13:19:52][C][ratgdo.binary_sensor:052]: RATGDO BinarySensor 'Motion'
[13:19:52][C][ratgdo.binary_sensor:052]:   Device Class: 'motion'
[13:19:52][C][ratgdo.binary_sensor:054]:   Type: Motion
[13:19:52][C][ratgdo.binary_sensor:052]: RATGDO BinarySensor 'Obstruction'
[13:19:52][C][ratgdo.binary_sensor:052]:   Device Class: 'problem'
[13:19:52][C][ratgdo.binary_sensor:056]:   Type: Obstruction
[13:19:52][C][ratgdo.binary_sensor:052]: RATGDO BinarySensor 'Button'
[13:19:52][C][ratgdo.binary_sensor:060]:   Type: Button
[13:19:52][C][ratgdo.binary_sensor:052]: RATGDO BinarySensor 'Motor'
[13:19:52][C][ratgdo.binary_sensor:052]:   Device Class: 'running'
[13:19:52][C][ratgdo.binary_sensor:058]:   Type: Motor
[13:19:52][C][ratgdo.cover:014]: RATGDO Cover 'Door'
[13:19:52][C][ratgdo.cover:014]:   Device Class: 'garage'
[13:19:52][C][ratgdo.light:014]: RATGDO Light
[13:19:52][C][restart.button:017]: Restart Button 'Restart'
[13:19:52][C][restart.button:017]:   Icon: 'mdi:restart'
[13:19:52][C][safe_mode.button:024]: Safe Mode Button 'Safe mode boot'
[13:19:52][C][safe_mode.button:024]:   Icon: 'mdi:restart-alert'
[13:19:52][C][homeassistant.time:010]: Home Assistant Time:
[13:19:52][C][homeassistant.time:011]:   Timezone: 'PST8PDT,M3.2.0,M11.1.0'
[13:19:52][C][web_server:153]: Web Server:
[13:19:52][C][web_server:154]:   Address: ratgdo.local:80
[13:19:52][C][mdns:116]: mDNS:
[13:19:52][C][mdns:117]:   Hostname: ratgdo
[13:19:52][C][esphome.ota:073]: Over-The-Air updates:
[13:19:52][C][esphome.ota:074]:   Address: ratgdo.local:8266
[13:19:52][C][esphome.ota:075]:   Version: 2
[13:19:52][C][safe_mode:018]: Safe Mode:
[13:19:52][C][safe_mode:019]:   Boot considered successful after 60 seconds
[13:19:52][C][safe_mode:021]:   Invoke after 10 boot attempts
[13:19:52][C][safe_mode:022]:   Remain in safe mode for 300 seconds
[13:19:52][C][api:140]: API Server:
[13:19:52][C][api:141]:   Address: ratgdo.local:6053
[13:19:52][C][api:143]:   Using noise encryption: YES
[13:19:52][C][improv_serial:032]: Improv Serial:
[13:21:18][D][time:050]: Synchronized time: 2025-02-25 13:21:18
[13:21:18][D][time:050]: Synchronized time: 2025-02-25 13:21:18
[13:22:06][D][button:010]: 'Sync' Pressed.
[13:22:09][D][button:010]: 'Sync' Pressed.
[13:22:20][D][button:010]: 'Query openings' Pressed.
[13:22:20][D][number:012]: 'Rolling code counter': Sending state 14146.000000
[13:22:20][D][ratgdo:288]: Openings: 7910
[13:23:10][D][esp8266.preferences:238]: Saving preferences to flash...
[13:23:28][D][button:010]: 'Query openings' Pressed.
[13:23:28][D][number:012]: 'Rolling code counter': Sending state 14147.000000
[13:23:28][D][ratgdo:288]: Openings: 7910
[13:23:39][D][button:010]: 'Query status' Pressed.
[13:23:39][D][number:012]: 'Rolling code counter': Sending state 14148.000000
[13:23:39][D][ratgdo:113]: Door state=CLOSED
[13:23:39][D][ratgdo:222]: Light state=OFF
[13:23:39][D][ratgdo:228]: Lock state=UNLOCKED
[13:23:39][D][ratgdo:207]: Learn state=INACTIVE
[13:23:43][D][button:010]: 'Query status' Pressed.
[13:23:43][D][number:012]: 'Rolling code counter': Sending state 14149.000000
[13:23:43][D][ratgdo:113]: Door state=CLOSED
[13:23:43][D][ratgdo:222]: Light state=OFF
[13:23:43][D][ratgdo:228]: Lock state=UNLOCKED
[13:23:43][D][ratgdo:207]: Learn state=INACTIVE
[13:24:10][D][esp8266.preferences:238]: Saving preferences to flash...
[13:24:28][D][ratgdo:113]: Door state=CLOSED
[13:24:28][D][ratgdo:222]: Light state=OFF
[13:24:28][D][ratgdo:228]: Lock state=UNLOCKED
[13:24:28][D][ratgdo:207]: Learn state=INACTIVE
```

csobrinho, 2025-02-26:
So a few more updates:
- from fully closed to `Position: 30%` seems to work (15:09:50) although I didn't measure the ratio of the opened door.
- from current 30% to 10% it it fully closed the door (15:10:16)
- from fully closed to 30% seems to have worked although (15:10:37)
- then 30% to fully open worked (15:10:45)
- then from fully open to 20% failed and completely closed the door (15:11:25)

Thanks for the help.

```
INFO ESPHome 2025.2.0
INFO Reading configuration /config/ratgdo-ca08fd.yaml...
INFO Updating https://github.com/ratgdo/esphome-ratgdo.git@main
INFO Updating https://github.com/ratgdo/esphome-ratgdo@None
INFO Updating https://github.com/ratgdo/esphome-ratgdo@main
INFO Detected timezone 'America/Los_Angeles'
INFO Starting log output from 10.10.1.28 using esphome API
INFO Successfully connected to ratgdo @ 10.10.1.28 in 0.006s
INFO Successful handshake with ratgdo @ 10.10.1.28 in 2.386s
[15:07:54][I][app:100]: ESPHome version 2025.2.0 compiled on Feb 25 2025, 11:53:26
[15:07:54][I][app:102]: Project ratgdo.esphome version 2.5i
[15:07:54][C][wifi:600]: WiFi:
[15:07:54][C][wifi:428]:   Local MAC: 48:3F:DA:CA:08:FD
[15:07:54][C][wifi:433]:   SSID: [redacted]
[15:07:54][C][wifi:436]:   IP Address: 10.10.1.28
[15:07:54][C][wifi:439]:   BSSID: [redacted]
[15:07:54][C][wifi:441]:   Hostname: 'ratgdo'
[15:07:54][C][wifi:443]:   Signal strength: -55 dB ▂▄▆█
[15:07:54][C][wifi:447]:   Channel: 1
[15:07:54][C][wifi:448]:   Subnet: 255.255.255.0
[15:07:54][C][wifi:449]:   Gateway: 10.10.1.1
[15:07:54][C][wifi:450]:   DNS1: 10.10.1.2
[15:07:54][C][wifi:451]:   DNS2: 10.10.1.1
[15:07:54][C][logger:177]: Logger:
[15:07:54][C][logger:178]:   Max Level: DEBUG
[15:07:54][C][logger:179]:   Initial Level: DEBUG
[15:07:54][C][logger:181]:   Log Baud Rate: 115200
[15:07:54][C][logger:182]:   Hardware UART: UART0
[15:07:54][C][ratgdo.number:024]: RATGDO Number 'Rolling code counter'
[15:07:54][C][ratgdo.number:024]:   Unit of Measurement: 'codes'
[15:07:54][C][ratgdo.number:028]:   Type: Rolling Code Counter
[15:07:54][C][ratgdo.number:024]: RATGDO Number 'Opening duration'
[15:07:54][C][ratgdo.number:024]:   Unit of Measurement: 's'
[15:07:54][C][ratgdo.number:030]:   Type: Opening Duration
[15:07:54][C][ratgdo.number:024]: RATGDO Number 'Closing duration'
[15:07:54][C][ratgdo.number:024]:   Unit of Measurement: 's'
[15:07:54][C][ratgdo.number:032]:   Type: Closing Duration
[15:07:54][C][ratgdo.number:024]: RATGDO Number 'Client ID'
[15:07:54][C][ratgdo.number:026]:  Type: Client ID
[15:07:54][C][switch.gpio:068]: GPIO Switch 'Status door'
[15:07:54][C][switch.gpio:090]:   Restore Mode: always OFF
[15:07:54][C][switch.gpio:031]:   Pin: GPIO16
[15:07:54][C][switch.gpio:068]: GPIO Switch 'Status obstruction'
[15:07:54][C][switch.gpio:090]:   Restore Mode: always OFF
[15:07:54][C][switch.gpio:031]:   Pin: GPIO15
[15:07:54][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Dry contact open'
[15:07:54][C][gpio.binary_sensor:016]:   Pin: GPIO14
[15:07:54][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Dry contact close'
[15:07:54][C][gpio.binary_sensor:016]:   Pin: GPIO12
[15:07:54][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Dry contact light'
[15:07:54][C][gpio.binary_sensor:016]:   Pin: GPIO0
[15:07:54][C][light:092]: Light 'Light'
[15:07:54][C][version.text_sensor:021]: Version Text Sensor 'Firmware Version'
[15:07:54][C][version.text_sensor:021]:   Icon: 'mdi:new-box'
[15:07:54][C][ratgdo:104]: Setting up RATGDO...
[15:07:54][C][ratgdo:105]:   Output GDO Pin: GPIO5
[15:07:54][C][ratgdo:106]:   Input GDO Pin: GPIO4
[15:07:54][C][ratgdo:107]:   Input Obstruction Pin: GPIO13
[15:07:54][C][ratgdo_secplus2:059]:   Rolling Code Counter: 14173
[15:07:54][C][ratgdo_secplus2:060]:   Client ID: 1488185
[... 354 lines cut ...]
[15:11:38][D][cover:173]:   Position: 14%
[15:11:38][D][cover:186]:   Current Operation: CLOSING
[15:11:38][D][ratgdo:246]: Motor: state=ON
[15:11:38][D][cover:170]: 'Door' - Publishing:
[15:11:38][D][cover:173]:   Position: 10%
[15:11:38][D][cover:186]:   Current Operation: CLOSING
[15:11:39][D][cover:170]: 'Door' - Publishing:
[15:11:39][D][cover:173]:   Position: 6%
[15:11:39][D][cover:186]:   Current Operation: CLOSING
[15:11:41][D][ratgdo:113]: Door state=CLOSED
[15:11:41][D][ratgdo:635]: Cancelling position callbacks
[15:11:41][D][ratgdo:222]: Light state=ON
[15:11:41][D][ratgdo:228]: Lock state=UNLOCKED
[15:11:41][D][ratgdo:207]: Learn state=INACTIVE
[15:11:41][D][binary_sensor:036]: 'Motor': Sending state OFF
[15:11:41][D][number:012]: 'Rolling code counter': Sending state 14184.000000
[15:11:41][D][cover:170]: 'Door' - Publishing:
[15:11:41][D][cover:173]:   Position: 0%
[15:11:41][D][cover:186]:   Current Operation: IDLE
[15:11:41][D][switch:016]: 'Status door' Turning OFF.
[15:11:41][D][switch:055]: 'Status door': Sending state OFF
[15:11:41][D][ratgdo:288]: Openings: 7922
[15:11:41][D][sensor:093]: 'Openings': Sending state 7922.00000 openings with 0 decimals of accuracy

```

SaberShip, 2025-03-18:
I am able to reproduce this as well.  Running ESPHome on Home Assistant
Official hardware V2.5i on ESPHome 2025.2.2 with a purple learn button (Security + 1.0)

When the cover state is "CLOSED" I'm able to set any percent I like without a problem.
If the cover state is "OPEN" at any percentage, then only fully opening to 100% or fully closing to 0% works.  Anything else results in the garage door moving towards the value selected followed by opening to 100%.

Even when manually pressing "Stop Cover" the garage door briefly halts before opening to 100%.  It only functions as expected when the previous state was "CLOSED"

csobrinho, 2025-03-18:
Yep, exactly the same. Thanks for checking @SaberShip. I think there was a regression because when I installed the first time the cover was working dead on. I'll try to run a bisect between the version I think I had when I installed it and now to see if we can figure out when it broke.