ratgdo/esphome-ratgdo #380: How can I send the door action signal for 500ms?

brandoncc, 2025-03-06:
I'm setting up a ratgdo32 to control a gate opener. My gate opener needs me to hold the button down for 500ms before it tells the gate to open or close.

I'm using Home Assistant, but I'm pretty new to it. Is there a way I can update the configuration to do what I need? I tried using ESP Home device builder to add to the yaml config, but I haven't figured out how to make it work.

Thank you for your hard work on this firmware, I appreciate it.

PaulWieland, 2025-03-06:
The dry contact control protocol holds the red/ctrl line low for [500ms](https://github.com/ratgdo/esphome-ratgdo/blob/1ca12528490716563a32354f3e459cc00cd1d7a8/components/ratgdo/dry_contact.cpp#L106) by default. I do think it would be good to make this duration customizable through the WebUI though.

brandoncc, 2025-03-08:
Closing because I can confirm that the default install worked for me. If you are interested in a PR, I'd still be happy to take a shot. I looked for a contribution guide, but didn't find one.

closed 2025-03-08 by brandoncc