ratgdo/esphome-ratgdo #382: How to secure Ratgdo v2.53i WebPage with ESPHome firmware?  Cannot find configuration page.

benze, 2025-03-12:
I just installed a Ratgdo v2.53i with the ESPHome firmware (2025.2.0 Feb 25 2025) for Security+ 1.0 and added it to HA.

From the docs, it says that:
> After the firmware is flashed, the ratgdo will reboot and attempt to connect to your wifi. Once this is succesfull, click “Visit Device” from the ESP Web Tools page and you will be presented with ratgdo’s configuration page.

However once I flashed with ESPHome firmware and connected to "Visit Device", there was no configuration page.  Only the Web controller page.  According to https://paulwieland.github.io/ratgdo/02_configuration.html, I should be able to set an OTA Password and Device Name, but how do I access this configuration page?

Additionally, how do I go about setting a static IP for the device if I so choose?  

Thanks,

Eric

PaulWieland, 2025-03-12:
You've installed the ESPHome firmware and are reading the directions for the MQTT firmware. 

ESPHome's documentation explains how to password protect the web server component. You can also just remove the webserver from the configuration.
https://esphome.io/components/web_server.html

closed 2025-03-12 by PaulWieland

benze, 2025-03-12:
@PaulWieland Ah - got it.  Thanks for the clarification.  It was unclear to me that the page was not for the ESPHome version. 

So if I understand correctly, the correct way to accomplish this with an ESPHome version is to take control of the device in HA, customize the YAML (based on the ESPHome docs) and reflash the firmware via HA?

benze, 2025-03-12:
> What's your order number?

#5231