ratgdo/esphome-ratgdo #384: obsruction clear keeps repeating in the log on home assistant esphome

wallyto, 2025-03-15:
just installed new ratgdov32. It is attached to old Chamberlain  with purple button. So is setup with security 1.
everything works in HA. can open and close, and toggle the light. 
The device keeps repeating obstruction problem/ clear and repeats continuously.
it is scrolling too fast to capture the details from the log.
firmware shows 2025.2.2 Mar 15 2025, 10:00:15

wallyto, 2025-03-15:
some logs.

INFO ESPHome 2025.2.2
INFO Reading configuration /config/esphome/ratgdov32-e31674.yaml...
INFO Updating https://github.com/ratgdo/esphome-ratgdo@main
INFO Updating https://github.com/ratgdo/esphome-ratgdo@main
WARNING GPIO2 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'America/Chicago'
INFO Starting log output from 10.0.0.215 using esphome API
INFO Successfully connected to ratgdov32-e31674 @ 10.0.0.215 in 0.025s
INFO Successful handshake with ratgdov32-e31674 @ 10.0.0.215 in 0.098s
[16:08:47][I][app:100]: ESPHome version 2025.2.2 compiled on Mar 15 2025, 15:56:31
[16:08:47][I][app:102]: Project ratgdo.esphome version 32.0
[16:08:47][C][wifi:600]: WiFi:
[16:08:47][C][wifi:428]:   Local MAC: [redacted]
[16:08:47][C][wifi:433]:   SSID: '[redacted]
[16:08:47][C][wifi:436]:   IP Address: 10.0.0.215
[16:08:47][C][wifi:440]:   BSSID: [redacted]
[16:08:47][C][wifi:441]:   Hostname: 'ratgdov32-e31674'
[16:08:47][C][wifi:443]:   Signal strength: -52 dB ▂▄▆█
[16:08:47][C][wifi:447]:   Channel: 5
[16:08:47][C][wifi:448]:   Subnet: 255.255.255.0
[16:08:47][C][wifi:449]:   Gateway: 10.0.0.1
[16:08:47][C][wifi:450]:   DNS1: 10.0.0.1
[16:08:47][C][wifi:451]:   DNS2: 0.0.0.0
[16:08:47][C][logger:177]: Logger:
[16:08:47][C][logger:178]:   Max Level: DEBUG
[16:08:47][C][logger:179]:   Initial Level: DEBUG
[16:08:47][C][logger:181]:   Log Baud Rate: 115200
[16:08:47][C][logger:182]:   Hardware UART: UART0
[16:08:47][C][ratgdo.number:024]: RATGDO Number 'Rolling code counter'
[16:08:47][C][ratgdo.number:024]:   Unit of Measurement: 'codes'
[16:08:47][C][ratgdo.number:028]:   Type: Rolling Code Counter
[16:08:47][C][ratgdo.number:024]: RATGDO Number 'Opening duration'
[16:08:47][C][ratgdo.number:024]:   Unit of Measurement: 's'
[16:08:47][C][ratgdo.number:030]:   Type: Opening Duration
[16:08:47][C][ratgdo.number:024]: RATGDO Number 'Closing duration'
[16:08:47][C][ratgdo.number:024]:   Unit of Measurement: 's'
[16:08:47][C][ratgdo.number:032]:   Type: Closing Duration
[16:08:47][C][ratgdo.number:024]: RATGDO Number 'Client ID'
[16:08:47][C][ratgdo.number:026]:  Type: Client ID
[16:08:47][C][switch.gpio:068]: GPIO Switch 'Status door'
[16:08:47][C][switch.gpio:091]:   Restore Mode: always OFF
[16:08:47][C][switch.gpio:031]:   Pin: GPIO26
[16:08:47][D][ratgdo:235]: Obstruction: state=OBSTRUCTED
[16:08:47][C][switch.gpio:068]: GPIO Switch 'Status obstruction'
[16:08:47][C][switch.gpio:091]:   Restore Mode: always OFF
[16:08:47][C][switch.gpio:031]:   Pin: GPIO25
[16:08:47][D][binary_sensor:036]: 'Obstruction': Sending state OFF
[16:08:47][D][switch:016]: 'Status obstruction' Turning OFF.
[16:08:47][D][switch:055]: 'Status obstruction': Sending state OFF
[16:08:47][D][binary_sensor:036]: 'Obstruction': Sending state ON
[16:08:47][D][switch:012]: 'Status obstruction' Turning ON.
[16:08:47][D][switch:055]: 'Status obstruction': Sending state ON
[16:08:47][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Dry contact open'
[16:08:47][C][gpio.binary_sensor:016]:   Pin: GPIO13
[16:08:47][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Dry contact close'
[16:08:47][C][gpio.binary_sensor:016]:   Pin: GPIO14
[16:08:47][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Dry contact light'
[16:08:47][C][gpio.binary_sensor:016]:   Pin: GPIO27
[16:08:47][C][light:092]: Light 'Light'
[16:08:47][C][version.text_sensor:021]: Version Text Sensor 'Firmware Version'
[16:08:47][C][version.text_sensor:021]:   Icon: 'mdi:new-box'
[16:08:47][C][ratgdo:104]: Setting up RATGDO...
[16:08:47][C][ratgdo:105]:   Output GDO Pin: GPIO17
[16:08:47][C][ratgdo:106]:   Input GDO Pin: GPIO21
[16:08:47][C][ratgdo:107]:   Input Obstruction Pin: GPIO4
[16:08:47][C][ratgdo_secplus1:045]:   Protocol: SEC+ v1
[... 114 lines cut ...]
[16:08:54][D][ratgdo:228]: Lock state=UNLOCKED
[16:08:54][D][ratgdo:113]: Door state=CLOSED
[16:08:54][D][ratgdo:222]: Light state=OFF
[16:08:54][D][ratgdo:228]: Lock state=UNLOCKED
[16:08:54][D][ratgdo:235]: Obstruction: state=OBSTRUCTED
[16:08:54][D][binary_sensor:036]: 'Obstruction': Sending state OFF
[16:08:54][D][switch:016]: 'Status obstruction' Turning OFF.
[16:08:54][D][switch:055]: 'Status obstruction': Sending state OFF
[16:08:54][D][binary_sensor:036]: 'Obstruction': Sending state ON
[16:08:54][D][switch:012]: 'Status obstruction' Turning ON.
[16:08:54][D][switch:055]: 'Status obstruction': Sending state ON
[16:08:55][D][ratgdo:222]: Light state=OFF
[16:08:55][D][ratgdo:228]: Lock state=UNLOCKED
[16:08:55][D][ratgdo:113]: Door state=CLOSED
[16:08:55][D][ratgdo:222]: Light state=OFF
[16:08:55][D][ratgdo:228]: Lock state=UNLOCKED
[16:08:55][D][ratgdo:235]: Obstruction: state=OBSTRUCTED
[16:08:55][D][binary_sensor:036]: 'Obstruction': Sending state OFF
[16:08:55][D][switch:016]: 'Status obstruction' Turning OFF.
[16:08:55][D][switch:055]: 'Status obstruction': Sending state OFF
[16:08:55][D][binary_sensor:036]: 'Obstruction': Sending state ON
[16:08:55][D][switch:012]: 'Status obstruction' Turning ON.
[16:08:55][D][switch:055]: 'Status obstruction': Sending state ON
[16:08:56][D][ratgdo:222]: Light state=OFF
[16:08:56][D][ratgdo:228]: Lock state=UNLOCKED

GelidusResearch, 2025-03-18:
Workaround: https://github.com/GelidusResearch/esphome-ratgdo/releases/tag/1.0.0

wallyto, 2025-03-21:
What is the procedure to update from the zip file?
Thanks

GelidusResearch, 2025-03-21:
@wallyto You would need to map the correct pins for that hardware using esphome builder.  You would point to it as an external component. You can use the guides I have to produced to create a usable YAML definition. You cannot use the VL53L4CX ToF with this implementation

```#GDO setup guide
# The VL53L1X ToF Sensor on RATGDO firmmare requires the GelidusResearch Repo
external_components:
  - source:
      type: git
      url: https://github.com/gelidusresearch/esphome-ratgdo.git
      ref: 1.0.0
    components: [ ratgdo ]
    refresh: 0s

esphome:
  name: gdo1
  friendly_name: gdo1
  project:
    name: ratgdo.esphome # Required for homebrige-ratgdo uses
    version: "1.0"  

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  baud_rate: 921600 #0 # Disable logging when running a ToF or DHT sensor on a v1 board otherwise set to 921600 (faster logging improves response time) 

# Enable Home Assistant API
api:
  encryption:
    key: "<your key>"  #Preserve this entry

ota:
  - platform: esphome # post 2024.5.x
    password: "<your password>"  # recommened but can be ommited for testing
  

wifi:
  power_save_mode: none
  ssid: !secret wifi_ssid          # Check the secrets file for the correct values 
  password: !secret wifi_password  # Ditto

  ap:
    ssid: "gdo1"
    password: !secret iot_captive_password

captive_portal:

web_server:

preferences:
  flash_write_interval: 600s      # Ardunio does not support wear leveling keep at 600+
                                  # Note: SecplusGDO uses IDF and supports wear leveling

substitutions:
  id_prefix: gdo1
  friendly_name: "GDO"
  uart_tx_pin: GPIO22           # J4 Pin 1 Red or 3 Blue to Opener Red CTRL terminal
  uart_rx_pin: GPIO21           # J4 Pin 1 Red or 3 Blue to Opener Red CTRL terminal
  input_obst_pin: GPIO23        # J4 Pin 4 Grey OBST 
  dry_contact_open_pin: GPIO18  # J4 Pin 6 Green
  dry_contact_close_pin: GPIO19 # J4 Pin 7 Blue
  dry_contact_light_pin: GPIO17 # J4 Pin 8 Orange
  tof_sda_pin: GPIO26           # GPIO26 = v2 board, GPIO3 = v1 board
  tof_scl_pin: GPIO25           # GPIO25 = v2 board, GPIO1 = v1 board

ratgdo:
  id: ${id_prefix}
  input_gdo_pin: ${uart_rx_pin}
[... 219 lines cut ...]
    on_press:
      then:
        lambda: !lambda |-
          id($id_prefix).query_openings();

  - platform: template
    id: ${id_prefix}_sync
    name: "Sync"
    entity_category: diagnostic
    on_press:
      then:
        lambda: !lambda |-
          id($id_prefix).sync();

  - platform: template
    id: ${id_prefix}_toggle_door
    name: "Toggle door"
    on_press:
      then:
        lambda: !lambda |-
          id($id_prefix).door_toggle();

status_led:
  pin: GPIO4
~~~