ratgdo/esphome-ratgdo #385: Feature Request: Force Close Door

CornHead764, 2025-03-17:
Obstruction sensors are an important safety device, however they can also be a nuisance, and often are in my case. Sometimes, despite nothing being in the way and the sensors being properly aligned, the garage door won't close, stating that there's an obstruction. The button on the wall provides the ability to override the obstruction sensors and forcefully close the door by holding down the button until the door is closed. Could we get this as a feature on the ratgdo? There are times where we're running late, pull out of the garage, hit close and it refuses to close, and the only way to work around this is to force close it from the button on the wall, then exit a different door so you don't have to re-open the garage later. It would be mighty convenient to have an emergency override button on my phone that I could press that would allow the garage to close while ignoring the obstruction sensors.

hotleadsingerguy, 2025-03-17:
Couldn't agree more. I have an old Genie H4000A with only dry contacts and obstruction sensors that are substantially too high voltage, which means I can't use them through the RATGDO. The problem is I can't even fake it...the RATGDO thinks the door is open and won't do a thing because it thinks there's an obstruction. Without a way around it I can't find a way to make this work.

PaulWieland, 2025-03-17:
@hotleadsingerguy this is not correct at all. The obstruction status reporting has absolutely nothing to do with ratgdo's ability or willingness to send a door command.

PaulWieland, 2025-03-17:
1. @CornHead764 Even if there is a way to do this, I don't like the idea of enabling such a feature for safety reasons. Do you know why your obstruction sensors aren't working properly? Are they dirty? Is the sun shining on them? I think remediating the root cause is the best course of action here.

CornHead764, 2025-03-17:
> Can you mount the sensors farther apart so that the wall always casts a shadow on the sensor?

From my troubleshooting, the problem seems to be more about shading the beam itself rather than the sensor

breakingspell, 2025-05-11:
@CornHead764 Interesting how your obstruction sensors aren't more reliable, considering the lengths you've gone to (shading with sun shields, etc). Have you tried replacing the sensors or wiring? 

Rather than in the ratgdo firmware, I handle "force closing" via Home Assistant automation, tracking the limit and obstruction sensor entities. Works every time and takes a little load away from the device. My back camera has a good angle of the garage door, so I also use it as a cropped peephole on my garage HA card to visually monitor. 

The script uses several `if` and `until` checks to evaluate the sensor states at each step. If the obstruction sensor causes the door to rebound and reach the `Open` limit sensor, the automation tries to close it again. It only exits the automation successfully if the `Closed` limit sensor confirms that the door is in the position for more than 5 seconds.

kevdogg, 2025-09-28:
I would second this feature. Similar problem -- actually north facing garage.  At certain months of the year during sunset the sun will shine on obstruction sensors which requires a manual holding of the button to close the door.  I suppose however possibly building a box around the sensor bodies might solve the problem.