ratgdo/esphome-ratgdo #386: Can't get RATGDO v2.53i to work with Genie H4000A (Dry Contact)

hotleadsingerguy, 2025-03-17:
Don't have any pictures atm, but I installed the ESPHome version and HA finds it find. I wired up the physical button to the RATGDO 1+2 and, at first, wired up the obstruction sensors to 3+4 before seeing on the internet that my obstruction sensors are probably too high voltage (12v+). I unplugged the black wire from the RATGDO to the GDO and the first time I pressed the "Toggle" button it worked...and then never did again. I've tried bypassing the RATGDO with the obstruction sensors directly to the GDO, but nothing happens. The RAT sees the door as "OPEN" and has never changed. Pressing the "Open" button says it's already open and that it'll skip the commend, pressing the "Close" button just says "Limit 0%" and nothing else.

This is an old GDO, I understand that...but I'm fine completely bypassing the obstruction sensors, if I can, or just letting the GDO handle their use. Is there any way past this? I eventually am going to splice into the limit switches to have that functionality, but I wanted to make sure the basics worked, first.

PaulWieland, 2025-03-17:
What make/model garage door opener do you have? What control protocol are you using?

PaulWieland, 2025-03-17:
How is ratgdo supposed to know your dry contact door state if you don't have the limit switches connected? :)

hotleadsingerguy, 2025-03-17:
I'm fine with it not knowing the state - If you just Press the button (not the controls to Lower or Raise the door) it should just provide the dry contact regardless, no? So if I don't set up the limit switch connectivity there's no way to tell the RATGDO to make contact?

PaulWieland, 2025-03-17:
Without limit switches, the discrete open and close commands will not work. You have to use the `Toggle door` button.

[image]

hotleadsingerguy, 2025-03-17:
That's the one I'm saying I was using. It does nothing and just says "Pressed Toggle" in the log

hotleadsingerguy, 2025-03-17:
Please don't close this - There's literally no other place or way for me to get help with the $66 device I bought, and following the instructions exactly as they're listed isn't allowing the device to work.

PaulWieland, 2025-03-17:
A photo of your wiring might help. The toggle door command sends a 500ms contact closure to the input on the door opener. If its not responding, then the most likely culprit is that something isn't wired correctly.

hotleadsingerguy, 2025-03-17:
BTW the only reason I have no pictures is because I had to un-do everything so the garage door could function again, while I troubleshoot. Should the device work with the sensors plugged directly into the GDO, or does it not function if they're not plugged in through the RATGDO? If it only works with the latter then I'm a bit out of luck, because nothing on the information or ordering page said it doesn't work with 12v+ Obstruction Sensors, but the old Genie ones seem to be.

The Genie H4000 has 4 connections, non color-coded, numbered 1-4. I have the Red+White plugged into 1+2, respectively, with the positive+negative from the button going into the first R+W on the RATGDO. I've tried the sensors combined into the W+B and the Black going to terminal #4, but because they're 12v I now know that won't work anyway.

I can re-do the installation later tonight and take pictures, but for now it prevents everything from working, so I can't keep it connected until I get it sorted.

PaulWieland, 2025-03-17:
The ratgdo obstruction sensor input is completely optional. Leave them connected to the garage door opener as they were originally and do not use the black/obst wire on the ratgdo wiring harness.

Your wall mounted control panel has two wires, one is ground, the other is ~12vdc. Connect the white wire to your common/ground terminal and the red wire to the wall control's + terminal.

hotleadsingerguy, 2025-03-17:
[image]
[image]

This is how it was configured. Worked the first press and then not again. Neither the button or handheld remote work. When I get home I can move the button ground directly to the GDO.

PaulWieland, 2025-03-17:
Did you check the polarity to ensure that terminal 2 is Ground?

hotleadsingerguy, 2025-03-17:
I didn’t yet. I can try flipping them tonight and grounding the wire to the terminal.

PaulWieland, 2025-03-17:
You can’t just hook it up backwards. If you don’t have a multimeter, look at your manual or at the terminal labels on the wall control to try and figure out which one is ground.

hotleadsingerguy, 2025-03-18:
Well I can't understand why, but the technical manual for the H4000 doesn't match the actual unit, and I never would have known because the old buttons were connected backwards (at least by the wiring). The buttons don't have a positive and negative, but the wiring used had the "positive" side hooked up to the neutral on the terminal.

Multimeter showed .5v going between the #1 and #2 (should have been positive+neutral, ended up being neutral+positive), 10v going between #2 and #3 (should both be neutral) and -10v going #3 to #4 (obstruction sensors, Neutral+Ground).

None of it makes sense, but flipping things around so the positive charge was going through the Red to White fixed it. Obstruction Sensors are connected directly to the GDO terminals and only R+W are going to the GDO from the RATGDO. Everything's working fine, at least until I hook up the limit wires. Thanks for the help!

hotleadsingerguy, 2025-03-18:
I guess one last question, though this issue is closed...for the limit switches can I just hook up the ground for it (pin 1 on the 6-pin cable) to a common ground on the unit, since the other alternative is accessing inside of the unit to find where it grounds the switches?

closed 2025-03-18 by hotleadsingerguy