ratgdo/esphome-ratgdo #389: Feature request: configure secplusv1 wall panel emulation

viperidae, 2025-03-23:
I have garage door opener with no panel.

It would be nice to have a setting to skip the wall panel detection and go straight to emulation mode, skipping the hard coded 35 second timeout period on start up.

Specifically in the wall_panel_emulation function in secplus1.cpp

jdranchman, 2025-03-23:
I like @viperidae  's idea. A config option in the firmware to not have a wall control is interesting. I'd be ok living with a battery remote as a wall button to rid myself of the wall control sync/power issue. It better yet, a ratgdo wall button (hint hint) with it's own comm, pir, etc to the disco.

PaulWieland, 2025-03-23:
If all you want is door activation then you can replace the wall control with a door bell button.