ratgdo/esphome-ratgdo #390: Publish target position

martijndierckx, 2025-03-23:
I would like to add my ratgdo to Homekit via MQTT (mqttthing).
And it works ... but it would work even better if Ratgdo exposed the target position state, alongside the current position.

Thanks
Martijn

martijndierckx, 2025-04-03:
To be clear, I'm using ratgdo in a secplus2 environment. Not with dry contacts

breakingspell, 2025-04-03:
Exposing the target position would be useful for a few things in HA as well. 

I looked at duplicating the `current_position` attribute to map `target_position` to, but couldn't find the definition at first glance. I think it may be part of the upstream esphome source.

martijndierckx, 2025-04-04:
In a different cover setup I can achieve this by using open_action, close_action and stop_action. But these are not supported by the ratgdo platform:

`
globals:
  - id: cover_target_pos
    type: int
    restore_value: no
    initial_value: "0"

script:
  - id: update_target_position
    mode: restart
    parameters:
      new_target: int
    then:
      - lambda: |-
          if (id(cover_target_pos) != new_target) {  // Only update if value has changed
            id(cover_target_pos) = new_target;
            id(cover_target_position).update();  // Force immediate sensor update
          }

cover:
  - platform: current_based
    name: Cover
    id: rollerblind
    open_action:
      - switch.turn_on: ${open_relay}
      - lambda: |-
          id(update_target_position).execute(100);  // Target = fully open (100%)
    close_action:
      - switch.turn_on: ${close_relay}
      - lambda: |-
          id(update_target_position).execute(0);  // Target = fully closed (0%)
    stop_action:
      - switch.turn_off: r0
      - switch.turn_off: r1
      - lambda: |-
          id(update_target_position).execute(id(rollerblind).position * 100);  // Target = current position
`

PaulWieland, 2025-04-04:
Can't you just use this?

```
binary_sensor:
  - platform: template
    name: "Target Door State"
    id: ${id_prefix}_target_door_state
    lambda: |-
      if (id(${id_prefix}_garage_door).current_operation == CoverOperation::COVER_OPERATION_IDLE) {
        // Cover is idle
        if (id(${id_prefix}_garage_door).position == COVER_CLOSED) {
          // Cover is closed
          return 1;
        } else {
          // Cover is not closed
          return 0;
        }
      } else if (id(${id_prefix}_garage_door).current_operation == CoverOperation::COVER_OPERATION_OPENING) {
        // Cover is currently opening
        return 0;
      } else if (id(${id_prefix}_garage_door).current_operation == CoverOperation::COVER_OPERATION_CLOSING) {
        // Cover is currently closing
        return 1;
      }
```

martijndierckx, 2025-04-07:
I added it as shown below. Seems to do the trick:

```
esphome:
  ...
  on_boot:
    then:
      - delay: 5s  # Allow time for cover component to initialize
      - script.execute: ${id_prefix}_update_target_door_state

globals:
  - id: ${id_prefix}_door_target_state
    type: std::string
    restore_value: true

script:
  - id: ${id_prefix}_update_target_door_state
    then:
      - lambda: |-
          std::string new_state = "closed";
          
          if (id(${id_prefix}_garage_door).current_operation == cover::COVER_OPERATION_IDLE) {
            // Cover is idle
            if (id(${id_prefix}_garage_door).position == COVER_CLOSED) {
              // Cover is closed
              new_state = "closed";
            } else {
              // Cover is not closed
              new_state = "open";
            }
          } else if (id(${id_prefix}_garage_door).current_operation == cover::COVER_OPERATION_OPENING) {
            // Cover is currently opening
            new_state = "open";
          } else if (id(${id_prefix}_garage_door).current_operation == cover::COVER_OPERATION_CLOSING) {
            // Cover is currently closing
            new_state = "closed";
          }
          
          id(${id_prefix}_door_target_state) = new_state;         
      - mqtt.publish:
          topic: ratgdo/cover/door/target_state
          payload: !lambda 'return id(${id_prefix}_door_target_state);'

cover:
  - id: !extend ${id_prefix}_garage_door
    on_state_change:
      then:
        - script.execute: ${id_prefix}_update_target_door_state
```

PaulWieland, 2025-04-07:
I used the binary sensor because the states that HK requires for TargetDoorState are `0|1`. I suppose it could also be a number, but the target door states generated by the template + lambda should satisfy the requirement.